pcb2gcode/pcb2gcode #2: pcb2gcode windows release

phragment, 2015-03-25:
I tried to run pcb2gcode on Windows today, but it failed due to missing DLLs:
libcairo-2.dll  Which I could get from the GTK+ bundle.
libcairomm-1.0-1.dll  Which I didn't found prebuilt. (Only on dubious sites, which nobody should use.)
I would suggest to distribute them bundled with pcb2gcode or even link them staticallyfor Windows.

Corna, 2015-03-28:
Sorry! I've updated both the win32 binaries for pcb2gcode and pcb2gcodeGUI, now they works.
Unfortunately the resulting size is very huge (due to the included DLLs), and I'm trying to figure out how to statically build it in MSYS2.

Corna, 2015-04-10:
I can't find a way to statically build gtk2 on Windows, I don't think it's possible

closed 2015-04-10 by Corna

koppi, 2015-06-10:
@Corna did you have a look into http://mxe.cc/ ?

Corna, 2015-06-10:
Currently the easiest way to build pcb2gcode is [MSYS2](http://sourceforge.net/projects/msys2/), and it is *very* easy (install it, download the required libraries with pacman, libtoolize && aclocal && automake --add-missing && autoreconf && ./configure --prefix=/mingw32 && make)
I don't think that gtk2 can be built statically on windows