pcb2gcode/pcb2gcode #3: issues parsing supplied configuration file

patkan, 2015-03-28:
When starting pcb2gcode (I compiled current master) without arguments the following error comes:
```
Error parsing configuration file "millproject": unrecognised option 'zero-start'
Error: Safety height not specified.
```
While the second one is okay the first one should not be there. If I comment out the offending line in millproject, more options show up as conflicting:
```
zero-start
nog81
bridgesnum
zbridges
al-back
al-front
al-probefeed
al-x
al-y
software
```

Corna, 2015-03-28:
I don't have this issue:

    Nicola@Nicola-PC ~/winhome/Desktop/pcb2gcode32msys2
    $ ./pcb2gcode.exe
    Importing front side... not specified.
    Importing back side... not specified.
    Importing outline... not specified.
    Internal Error: No layers prepared.
    Importing drill... not specified.
    END.

Those options are the options I added in the last 3/4 months, are you sure you've compiled the official master?
Moreover the software should find the millproject file and find the corresponding "zsafe" line

patkan, 2015-03-28:
I forked pcb2gcode/pcb2gcode to patkan/pcb2gcode-1. Then I cloned the fork to my machine and made sure to check out master. My machine is a Linux-machine.
I built the project with `./git-build.sh`. But according to `pcb2gcode --version` I'm on version 1.1.4.

Corna, 2015-03-28:
git-build.sh only builds it, it does not execute a _make install_
Try

    make install
    pcb2gcode --version

or

    ./pcb2gcode --version

patkan, 2015-03-28:
I'm deeply sorry. :-/ Today seems not to be my day...

closed 2015-03-28 by patkan

Corna, 2015-03-28:
I made the same error a couple of days ago, don't worry! ;-)