pcb2gcode/pcb2gcode #4: 2 sided board issues

zoomzoomluke, 2015-03-28:
Hi,

Thanks for your excellent work.  I am however having a bit of a problem with producing double sided boards. Attached is my mill project file and some screenshots.
Basically when the AL probing happens on the back layer, the probing occurs in the incorrect place.

I'm not sure but I think that there may also be an issue with the units? The drilling shows different dimensions to the engraving.

[image]
[image]
[image]
[image]

millproject:
# Common options
dpi=1000
metric=true
metricoutput=true
mirror-absolute=true
optimise=true
zchange=10.0000
zero-start=false
zsafe=2.0000

# Mill options
extra-passes=0
mill-feed=600
mill-speed=10000
offset=0.1
#offset=25  # Uncomment (and comment the previous line) for voronoi regions
zwork=-0.085

# Drill options
drill-feed=600
drill-front=true
drill-speed=10000
milldrill=false
nog81=false
onedrill=true
zdrill=-1.65

# Outline options
bridges=1.5000
bridgesnum=2
cut-feed=400
cut-front=false
cut-infeed=0.4
cut-speed=10000
cutter-diameter=0.8
fill-outline=true
outline-width=0.2
zbridges=-0.8
zcut=-1.65

# Autoleveller options
al-back=true
al-front=true
al-probefeed=300
al-x=10.0000
al-y=10.0000
software=LinuxCNC
al-probe-on=(MSG, Ensure spindle if OFF and datum has been set before pressing "S <resume>")@M0
al-probe-off=(MSG, Switch spindle to ON position and speed at 100% before pressing "S <resume>")@M0

zoomzoomluke, 2015-03-28:
Here are the project files:
https://www.dropbox.com/s/lch4x3tmol7479h/project_files.zip?dl=0

Corna, 2015-03-28:
Which branch are you using?

zoomzoomluke, 2015-03-28:
currently on master

Corna, 2015-03-30:
If you open _drill.drd_ in text mode you'll see a _M72_ line: according to the [official excellon specifications](http://www.excellon.com/manuals/program.htm), _M72_ means "Inch Measuring Mode", so I suppose that your CAD software is exporting the gerber files in metric mode and the drill file is imperial mode.

Corna, 2015-03-30:
With the new autoleveller version it works in my simulator. Try it

zoomzoomluke, 2015-03-30:
Great that worked fine.

By default eagle outputs all gerbers and drills imperial. It's difficult to change this without hacking the eagle.def file.

Perhaps an option --x-format=imperial or --x-format=metric could be implemented? This would allow for different measuring mode inputs than the program's working units? 

Oh and just another strange problem. The milldrill option is resulting in the following error.
[image]

Thanks so much for your help :)

Corna, 2015-03-31:
Here's a screenshot of your front + drill in the latest (git) gerbv
[image]
As you can see gerbv fails to detect the real size of your drill file
Can you please check that the exported drill file is correct?

The milldrill bug is known, the milldrill option has always been broken, maybe I'll fix it after the work on autoleveller

zoomzoomluke, 2015-03-31:
All working now, thanks. Turns out eagle 7.2+ made a change to the CAM processor and needed updated files.

Thanks for all your help.

closed 2015-03-31 by zoomzoomluke

Corna, 2015-04-02:
Btw, milldrill functionality has been fixed in commit 9d3785b2b067075a6ecbae24ee3a7c5929acf18a