pcb2gcode/pcb2gcode #5: grbl support for autoleveler

maleadt, 2015-04-21:
I wondered whether it would be possible to extend the autoleveler such that it also supports the grbl g-code interpreter. The major issue seems the lack of variable support (see grbl/grbl#413), so I'm not sure whether it is even doable.

maleadt, 2015-04-21:
bCNC apparently supports height mapping with grbl's probing functionality, but then again it does this by taking care of sending the g-code itself, allowing to parse its output.

Corna, 2015-04-22:
Unfortunately it's not possible. As you pointed, grbl does not supports variables, and variables are the key feature required for the autoleveller. The problem is that the initial gcode "does not know" the heigth of the probe points, and the only possible implementation is a parametric gcode, unsupported in grbl.
bCNC is basically a grbl interface and, I suppose, it bypasses this problem saving each height in the PC and computing the corrected height directly in the PC. If you think that an implementation is possible, or if grbl suddenly starts supporting variables, reoen this issue.

closed 2015-04-22 by Corna