pcb2gcode/pcb2gcode #6: Separate feedrate for penetration (vertical) and milling (horizontal)

kajuk, 2015-04-23:
Is it possible to add support for separate feedrate values for penetration (vertical) and milling (horizontal) moves during isolation milling and edge cutting?

Corna, 2015-04-23:
Well, it's easy doable... But is it really necessary? We already have a huge amount of command line arguments
Why do you need it?

kajuk, 2015-04-23:
There reason I ask for it is that my V bits can easily sustain milling feedrate of 800mm/m and even higher but their tips also break easily when the feedrate is above ~150-200mm/m
With the current pcb2gcode I have to apply some regex replacement in produced gcode to insert lower feedrate before plunge and restore higher feedrate after that.

Corna, 2015-04-23:
You're right, It's a good reason
I've added the command line options _--mill-vertfeed_ and _--cut-vertfeed_ in commit 2881e43bab127e7cc4cd45650a3ee609fc4bd4c5
If unspecified, _--mill-vertfeed_ / 2 and _--cut-vertfeed_ / 2 will be used
Can you please test it?

kajuk, 2015-04-23:
Many thanks! I will test it tonight.

kajuk, 2015-04-23:
Works just as intended, thanks you! I'm sure you meant --mill-feed/2 and --cut-feed/2 in your previous post.

Corna, 2015-04-23:
Yes, it was a typo, fixed

closed 2015-04-23 by Corna