pcb2gcode/pcb2gcode #7: When --outline is specified, gcode for copper milling is empty

kajuk, 2015-04-23:
I don't know if it's expected behavior or not but when I specify --outline for board cutting at the same time as --back and --drill, the copper milling gcode is mostly empty.
E.g. with the command 'pcb2gcode --drill xhc-hb04-keyboard.drl --back xhc-hb04-keyboard-B_Cu.gbr --basename xhc-keyboard --outline xhc-hb04-keyboard-Edge_Cuts.gbr' I get in xhc-keyboard_back.ngc:

( pcb2gcode 1.2.1 )
( Software-independent Gcode )
      
G94 ( Millimeters per minute feed rate. )
G21 ( Units == Millimeters. )

G90 ( Absolute coordinates. )
S18000 ( RPM spindle speed. )
G64 P0.05080 ( set maximum deviation from commanded toolpath )
F800.00000 ( Feedrate. )

F800.00000 ( Feedrate. )
M3 ( Spindle on clockwise. )

G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z10.00000 ( retract )

M5 ( Spindle off. )
M9 ( Coolant off. )
M2 ( Program end. )

If I do it without --outline I get:

( pcb2gcode 1.2.1 )
( Software-independent Gcode )
      
G94 ( Millimeters per minute feed rate. )
G21 ( Units == Millimeters. )

G90 ( Absolute coordinates. )
S18000 ( RPM spindle speed. )
G64 P0.05080 ( set maximum deviation from commanded toolpath )
F800.00000 ( Feedrate. )

F800.00000 ( Feedrate. )
M3 ( Spindle on clockwise. )
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z2.00000 ( retract )

G00 X9.15670 Y108.66120 ( rapid move to begin. )
F100.00000
G01 Z-0.10000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
F800.00000
X9.15670 Y108.66120
X9.13130 Y108.10240
X0.80010 Y99.77120
X0.74930 Y99.61880
X0.74930 Y61.11240
....
X20.53590 Y7.56920
X20.45970 Y7.39140

G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z10.00000 ( retract )

M5 ( Spindle off. )
M9 ( Coolant off. )
M2 ( Program end. )

Corna, 2015-04-23:
You probably need to specify _--fill-outline_, that tells the software to treat the outline file as the contour of a polyon. If unspecified, it treats it as a sequence of lines, and the resulting outline mask is a thin outline with a giant hole in it
Can you test it with fill-outline enabled?

kajuk, 2015-04-23:
With --fill-outline it fails and doesn't produce back milling gcode file at all:
pcb2gcode --drill xhc-hb04-keyboard.drl --back xhc-hb04-keyboard-B_Cu.gbr --basename xhc-keyboard --fill-outline --outline xhc-hb04-keyboard-Edge_Cuts.gbr 
Importing front side... not specified.
Importing back side... DONE.
Importing outline... DONE.
Internal Error: Shrinking the outline by half the line width came to a halt.
Importing drill... DONE.
Exporting drill... DONE.

Also, I'm not sure I explained it enough because you're suggesting me something for the outline file but the problem is that when I specify --outline, it is actually the BACK copper gcode which is produced almost empty and I believe --fill-outline doesn't have anything to do with it.

Corna, 2015-04-24:
The outline file is also a mask for the other layers, that's why you have an empty back
Can you please email me your gerber files and your millproject file?

kajuk, 2015-04-24:
Hi Nicola,

I've attached gerber/millproject files.

Regards
Dmitri

Corna, 2015-04-24:
Email attachments are unsupported in github
Send the files directly to me, at nicola@corna.info

Corna, 2015-04-24:
Everything works fine with a larger value (like 0.1) of outline-width
You can't (yet) set outline-width to 0 or too close to 0
Can you confirm it?

kajuk, 2015-04-24:
Indeed, with outline-width 0.06 and higher it works although I fail to understand why it has affect on the back milling. May be it's worth giving a warning when specifying a too low outline-width?  
And thank you again!

Corna, 2015-04-24:
Yes, the next thing in my schedule is to fix outline-width
The outline files masks all the other layers; look at this example (I've made the outline lines thicker just to explain the problem better)

[image]

Here's the result with fill-outline enabled

[image]
[image]
[image]
[image]

As you can see the back layer is masked with the filled outline polygon
This is the result without fill-outline

[image]
[image]
[image]

With fill-outline turned off the "considered" area is just the border (that usually is 0.1 mm thick) and not the area within the rectangle; since the tracks doesn't usually overlap the border, the resulting "valid area" is empty
fill-outline is the most common behaviour, I'm thinking to set it enabled by default

closed 2015-04-24 by Corna

eyal0, 2016-11-10:
I'm still having this problem, even when I use --fill-outline.  Can you take a look?  I tried debugging it but I can't find the issue.  My gerber files are attached and the command line that I ran is:

pcb2gcode --noconfigfile --back multivibrator-B.Cu.gbr --drill multivibrator.drl --outline multivibrator-Edge.Cuts.gbr --output-dir=/home/eyalsoha/private/kicad/multivibrator/cnc --mirror-absolute=false --tile-x=1 --tile-y=1 --tolerance=0.001 --vectorial=true --zchange=10.0000 --zero-start=true --zsafe=1.0000 --mill-feed=360 --mill-speed=12000 --voronoi=true --zwork=-0.04 --drill-feed=30 --drill-side=auto --drill-speed=12000 --milldrill=true --onedrill=true --zdrill=-1.6 --cut-feed=100 --cut-infeed=100 --cut-side=auto --cut-speed=10000 --cutter-diameter=0.8 --milldrill-diameter=0.8 --zbridges=-0.6000 --zcut=-0.13 --fill-outline

[multivibrator.zip](https://github.com/pcb2gcode/pcb2gcode/files/584399/multivibrator.zip)

Let me know if you need more information from me.