pcb2gcode/pcb2gcode #8: Feature request: x,y move before z

kajuk, 2015-04-23:
Is it possible to add an option to make x,y move first before move to z safe?
At the moment first move is to z safe as below:

M3 ( Spindle on clockwise. )
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z2.00000 ( retract )

G00 X9.15670 Y108.66120 ( rapid move to begin. )

I use low safe z levels to minimize air cutting and reduce milling/drilling time. With the first move to safe z there is very big chance that the next x,y move may bump into fixtures if the spindle is not positioned within milling area before the start.

Corna, 2015-04-27:
I don't understand your problem: why do you think it's dangerous to move at zsafe?
Also the problem can be reversed: if the spindle is initially at a lower height than zsafe (for example, because it was previously milling and it was interrupted) a horizontal movement is very dangerous.

kajuk, 2015-04-27:
Because I use very low zsafe values my fixtures are higher than zsafe and if the spindle is outside of xy milling area at the start then the first move goes to zsafe level and then a move into xy area may cross one of the fixtures and break the tool.
I believe the first "xy move and then z move" to zsafe is safer because most of the time spindle is above milling area at the start of milling jobs.
At least it would be nice to have an option to choose first move: "xy and the z" or "z and then xy".

kajuk, 2015-04-27:
Or I'd suggest a way to satisfy both initial spindle positions. If you introduce something like zrapid option then:
1. The first move is to zsafe then xy move then to zrapid
2. All subsequent rapids are done at zrapid
3. The last move is to zsafe.

This way the tool will be safe in both high low and high initial spindle positions and all rapids will be done at low zrapid level to minimize milling time.

corna, 2015-12-16:
This issue is outdated and I still disagree about an additional "zrapid" argument. zsafe is a safety height where all the horizontal movements are made; if you want to reduce the execution time you can reduce the zsafe, but make sure that no fixtures are in the work area. Any different workflow is unsafe, and I prefer safety on speed.

closed 2015-12-16 by corna