pcb2gcode/pcb2gcode #11: no outline for holes

koppi, 2015-06-08:
I'm trying to create an arduino shield using Kicad and pcb2gcode, see: https://github.com/koppi/arduino-shield

pcb backside:
https://github.com/koppi/arduino-shield/blob/master/outp0_original_back.png

Somehow https://github.com/koppi/arduino-shield/blob/master/back.ngc does not contain the milling paths for isolating the pin raster.

Can you have a look into the Kicad project files? Am I missing any step there in the export? Or is there something wrong in https://github.com/koppi/arduino-shield/blob/master/millproject

koppi, 2015-06-08:
Oki, got it working: removed the millproject file, added a Makefile with the correct params: https://github.com/koppi/arduino-shield/blob/master/Makefile

closed 2015-06-08 by koppi

Corna, 2015-06-08:
Did you changed the parameters when you switched from the millproject file to the Makefile?

koppi, 2015-06-08:
yes