pcb2gcode/pcb2gcode #12: TSP path optimization missing

koppi, 2015-06-08:
I wrote a small Python script that optimizes the drill file's path, see:
https://github.com/curtacircuitos/pcb-tools/issues/30

It would be super cool, if pcb2gcode could support TSP path optimization.

Corna, 2015-06-10:
I'm working on a simple nearest neighbour implementation, and it works quite well right now
I'll shortly create a new test branch for it

koppi, 2015-06-10:
:thumbsup:

koppi, 2015-06-11:
please report back here, when it's ready to test.

Corna, 2015-06-11:
Here it is
commit e3c49955ae3eea3cc9e10d8c1b3c638e8eb7d733 (in the new branch [drill_optimisation](https://github.com/pcb2gcode/pcb2gcode/tree/drill_optimisation))
Read the commit description for more info

koppi, 2015-06-11:
I just tested it with my arduino-shield, works perfect, thanks a lot! :100:

Corna, 2015-06-11:
Can you please report the path before and after the optimisation?

Corna, 2015-06-11:
Can you please paste here the paths before and after the optimisation?

koppi, 2015-06-11:
Drill path without optimization:
[image]
Drill path with optimization:
[image]

Corna, 2015-06-11:
Thanks
Good, after some tests I'll merge it into the master branch

Corna, 2015-06-12:
Merged

closed 2015-06-12 by Corna

matthijskooijman, 2015-10-28:
Thanks for this change, it was on my lists of annoyances to fix as well :-D

corna, 2015-10-28:
You're welcome :-)