pcb2gcode/pcb2gcode #14: A bit more overlap in the milling path is needed

koppi, 2015-06-10:
Is it possible to add a bit of "overlap" between a milling path's start and end point? There are some bridges in my first try to mill a PCB, see:
[image]

https://github.com/koppi/arduino-shield

Corna, 2015-06-11:
I've never had this problem, but other users in the previous years have reported the same issue. I'll investigate

koppi, 2015-06-11:
I'm running this on LinuxCNC, is the "G64 P0.05080 ( set maximum deviation from commanded toolpath )" relevant here?

Corna, 2015-06-11:
From your Makefile I see that you're using metric units, therefore that G64 code allows a maximum deviation of 0.05 mm, not enough to create those errors

koppi, 2015-06-11:
While pcb2gcode currently outputs "mathematical correct" milling paths, it would be really helpful to have a bit of overlap between the current end and start points. I think, every machine has some backslash and tolerances, if you could add this as a feature I will test it.

Corna, 2015-06-12:
Can you please mill the same circuit in the same conditions with the optimisation turned off? I think that it's an optimisation bug

koppi, 2015-06-12:
Here are the two versions with and without optimise:
* [arduino-shield-no-opt_back.ngc](http://paste.ubuntu.com/11701615/)
* [arduino-shield-opt_back.ngc](http://paste.ubuntu.com/11701617/)
Whenn I CNC arduino-shield-no-opt_back.ngc, I get the correct results.

Corna, 2015-06-13:
It was a simplifypath bug
I've finally added Boost.Geometry and I've replaced the local geometric algorithms (pointDistance and simplifypath) with the Boost implementation (boost::geometry::distance and boost::geometry::simplify, that is Douglas-Peucker)
The milling files are now slightly larger, but they are much more precise (for example, circles with simplifypath were horrible) and this bug should now be fixed
Can you please pull the code from the new [boost_geometry](https://github.com/pcb2gcode/pcb2gcode/tree/boost_geometry) branch and test if the problem is still present with the optimisation turned on?

koppi, 2015-06-13:
Just tested it withe the boost_geometry branch, results look much better now.

Corna, 2015-06-14:
Perfect!
Thanks for your help

closed 2015-06-14 by Corna