pcb2gcode/pcb2gcode #15: Please add some gerbers for test milling

koppi, 2015-06-12:
I found some gerbers in ```testing/gerbv_example```, but what is missing IMHO is a gerber / kicad file for calibrating CNC process parameters.

I googled and found http://phk.freebsd.dk/CncPcb/calibrate.html maybe @Corna could you add some calibrating kicad / gerber files for reference in pcb2gcode?

Corna, 2015-06-15:
Good idea, I had hard time figuring how precise were my bits
A good calibration pattern should include
 * A minimum track width test
 * A minimum clearance test (based on the previous result)
 * A maximum feedrate test (based on the previous results)
 * A copper thickness test

Do you agree?

koppi, 2015-06-15:
sounds good.

goranskular, 2016-10-13:
I found [pcb-probe](https://github.com/atom-bomb/pcb-probe)

It seems it works well. See here: [using pcb-probe](http://www.instructables.com/id/Make-your-own-PCBs-on-an-inexpensive-desktop-CNC-m/l)

corna, 2016-10-13:
If I understood the article correctly, pcb-probe adds the probing routines on the gcode file, but this is already integrated in the pcb2gcode's module _autoleveller_
In this issue the request is to add some gcode files to obtain the parameters of the milling machine, like the real width of the milling bit or the width of the copper layer on the board

goranskular, 2016-10-14:
I have a strange problem with probing. Probably I am doing something wrong, but after the probing it seems that my results are rotated on Y axis. Here is the picture of the result, and probing file with gcode. That's why I wanted also to try the pcb-probe approach.. If I am doing something wrong, can you point me what? Thanks

[image]

Mach 3 gcode: [mach3gcode.zip](https://github.com/pcb2gcode/pcb2gcode/files/529574/mach3gcode.zip)

From probing: [probe.txt](https://github.com/pcb2gcode/pcb2gcode/files/529572/probe.txt)

goranskular, 2016-10-15:
Sorry for that. I had a problem with autolevelling not related to pcb2gcode, but it was related to a problem with USB motioncontroller. Probing was not working well, so I needed to add a macro before setting a variable from #2002 after probing. More here: [cnczone USB motion controller G31 solution](http://www.cnczone.com/forums/xzero-cnc/192666-cnc.html)

Thanks!

corna, 2016-10-15:
Great!
Is it possible to integrate the command _SetVar(#2002,getoemdro(802))_ directly in the gcode, without the need of an external m1s file?

corna, 2016-10-15:
I've **never** tested the autoleveller with Mach3, I just wrote it for LinuxCNC and then I translated all the codes following the Mach3 specifications: can you please provide me a full report of what works (and what doesn't) with Mach3?
Thanks

goranskular, 2016-10-15:
It seems it is not possible to do it without a macro. I will test everything I can and let you know. It seems everything is working as it should.

closed 2021-03-26 by koppi