pcb2gcode/pcb2gcode #16: Please add support for tiling the pcb x,y-times

koppi, 2015-06-15:
It would be super handy to mill more than one pcb during the fabrication process.

Please add support for tiling the pcb x and y times in x and y direction.

Corna, 2015-06-15:
Unfortunately this is not very easy to implement due to the huge number of differences between the cnc controllers... While it is quite easy with LinuxCNC (the entire project can be a subroutine that can be called multiple times with different coordinate system values), it is very hard with grbl, and the only possible solution is to concatenate the same code multiple times
But I realise that this is an useful feature, I'll try to implement it

koppi, 2015-06-15:
The solution with concatenation sounds ok to me. I was starting to hack two new options ```--tile-x``` and ```tile-y``` into ```options.cpp```, then hacked into ```ngc_exporter.cpp``` the tiling stuff, but I do not know howto add this feature into ```autoleveler.cpp```. Maybe you know better howto write code for all of this, I'll be happy to test it out then.

Corna, 2015-06-15:
You have to use subroutines if the user specify a software that supports them (LinuxCNC, Mach3 or Mach4), you can't just force the concatenation for every controller
It is not an easy enhancement, I'll do it in the following

koppi, 2015-06-15:
I'm using OpenSCAM to preview the ngc files. I'm using LinuxCNC to mill these files. I noticed, that OpenSCAM does not support subroutines, so usually I generate two versions of the ngc files: a version with probing and a version without probing. For the tiling feature, it would be ok for me, if the ngc files do not use additional subroutines but rather get bigger in size.

Corna, 2015-06-16:
Code concatenation is a really bad choice and I don't want to use it unless really necessary
Why don't you use linuxcnc-sim as a simulator?

koppi, 2015-06-19:
Did you already start implementing this feature request?

I just made a video showing the milling process: https://www.youtube.com/watch?v=2nGcliWefek

Corna, 2015-06-20:
Unfortunately I'm very busy these days, I'll resume my work on pcb2gcode in the next months

corna, 2015-11-16:
Hi
I've almost done it, are you available for testing?

corna, 2015-11-16:
Great! I just have to finish the drill part, then I'll push it in a separate branch
Do you use only LinuxCNC?

corna, 2015-11-16:
I'm not in a hurry, and I'll complete it in a few days

corna, 2015-11-19:
Here it is
https://github.com/corna/pcb2gcode/tree/travis_containers
I haven't updated the man yet; to use it just specify __--tile-x=[number of columns]__ and __--tile-y=[number of rows]__. Please test it both with __--software=LinuxCNC__ and __--software=Custom__.
Thanks

koppi, 2015-11-19:
I'm stuck at:
```
configure: error: cannot find the flags to link with Boost program_options
```
Ubuntu 15.10 here, do you have any hints? Here's the full log (created with ```script pcb2gcode.log```):

```
$ git clone https://github.com/corna/pcb2gcode
Cloning into 'pcb2gcode'...
remote: Counting objects: 1230, done.[K
remote: Compressing objects:   5% (1/20)   [K
remote: Compressing objects:  10% (2/20)   [K
remote: Compressing objects:  15% (3/20)   [K
remote: Compressing objects:  20% (4/20)   [K
remote: Compressing objects:  25% (5/20)   [K
remote: Compressing objects:  30% (6/20)   [K
remote: Compressing objects:  35% (7/20)   [K
remote: Compressing objects:  40% (8/20)   [K
remote: Compressing objects:  45% (9/20)   [K
remote: Compressing objects:  50% (10/20)   [K
remote: Compressing objects:  55% (11/20)   [K
remote: Compressing objects:  60% (12/20)   [K
remote: Compressing objects:  65% (13/20)   [K
remote: Compressing objects:  70% (14/20)   [K
remote: Compressing objects:  75% (15/20)   [K
remote: Compressing objects:  80% (16/20)   [K
remote: Compressing objects:  85% (17/20)   [K
remote: Compressing objects:  90% (18/20)   [K
remote: Compressing objects:  95% (19/20)   [K
remote: Compressing objects: 100% (20/20)   [K
remote: Compressing objects: 100% (20/20), done.[K
Receiving objects:   0% (1/1230)   
Receiving objects:   1% (13/1230)   
Receiving objects:   2% (25/1230)   
Receiving objects:   3% (37/1230)   
Receiving objects:   4% (50/1230)   
Receiving objects:   5% (62/1230)   
Receiving objects:   6% (74/1230)   
Receiving objects:   7% (87/1230)   
Receiving objects:   8% (99/1230)   
Receiving objects:   9% (111/1230)   
Receiving objects:  10% (123/1230)   
Receiving objects:  11% (136/1230)   
Receiving objects:  12% (148/1230)   
Receiving objects:  13% (160/1230)   
Receiving objects:  14% (173/1230)   
Receiving objects:  15% (185/1230)   
Receiving objects:  16% (197/1230)   
Receiving objects:  17% (210/1230)   
Receiving objects:  18% (222/1230)   
Receiving objects:  19% (234/1230)   
Receiving objects:  20% (246/1230)   
Receiving objects:  21% (259/1230)   
Receiving objects:  22% (271/1230)   
Receiving objects:  23% (283/1230)   
Receiving objects:  24% (296/1230)   
Receiving objects:  25% (308/1230)   
Receiving objects:  26% (320/1230)   
Receiving objects:  27% (333/1230)   
Receiving objects:  28% (345/1230)   
Receiving objects:  29% (357/1230)   
Receiving objects:  30% (369/1230)   
Receiving objects:  31% (382/1230)   
Receiving objects:  32% (394/1230)   
Receiving objects:  33% (406/1230)   
Receiving objects:  34% (419/1230), 436.01 KiB | 850.00 KiB/s   
Receiving objects:  35% (431/1230), 436.01 KiB | 850.00 KiB/s   
Receiving objects:  36% (443/1230), 436.01 KiB | 850.00 KiB/s   
Receiving objects:  37% (456/1230), 436.01 KiB | 850.00 KiB/s   
Receiving objects:  38% (468/1230), 436.01 KiB | 850.00 KiB/s   
[... 211 lines cut ...]
checking how to run the C++ preprocessor... g++ -E
checking for ld used by g++... /usr/bin/ld -m elf_x86_64
checking if the linker (/usr/bin/ld -m elf_x86_64) is GNU ld... yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking for g++ option to produce PIC... -fPIC -DPIC
checking if g++ PIC flag -fPIC -DPIC works... yes
checking if g++ static flag -static works... yes
checking if g++ supports -c -o file.o... yes
checking if g++ supports -c -o file.o... (cached) yes
checking whether the g++ linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking for Boost headers version >= 1.47.0... yes
checking for Boost's header version... 1_58
checking for the toolset name used by Boost for g++... configure: WARNING: could not figure out which toolset name to use for g++

checking boost/program_options.hpp usability... yes
checking boost/program_options.hpp presence... yes
checking for boost/program_options.hpp... yes
checking for the Boost program_options library... no
configure: error: cannot find the flags to link with Boost program_options
[01;32mkoppi@x220[00m:[01;34m~/tmp/pcb2gcode[00m$ exit

Script done on Do 19 Nov 2015 16:33:12 CET
```

koppi, 2015-11-19:
Update: I just fixed this by installing the package ```libboost-program-options1.58-dev```.

koppi, 2015-11-19:
I added to the Makefile at https://github.com/koppi/arduino-shield :

```
--tile-x=1 --tile-y=2
```
and got a segmentation fault:
```
koppi@x220:~/Dropbox/cnc/arduino-shield$ make
#./drl-optimize.py arduino-shield.drl > arduino-shield.opt.drl
# without probing
pcb2gcode --tile-x=1 --tile-y=2 --basename=ngc/arduino-shield --drill=arduino-shield.drl --front=arduino-shield-F_Cu.gbr --back=arduino-shield-B_Cu.gbr --outline=arduino-shield-Cmts_User.gbr --output-dir=. --postamble=ngc/postamble.ngc --preamble=ngc/preamble.ngc --dpi=1000 --metric=true --metricoutput=true --mirror-absolute=false --optimise=true --zchange=40.0000 --zero-start=true --zsafe=0.5000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=1.0 --zwork=-0.075 --drill-feed=1000 --drill-front=false --drill-speed=20000 --milldrill=false --nog81=false --onedrill=true --zdrill=-2.0 --cut-front=true --cut-feed=600 --cut-infeed=10.0000 --cut-speed=10000 --cutter-diameter=2.0000 --fill-outline=true --outline-width=1 --zcut=-1.6500 --al-back=false --al-front=false --al-probefeed=100 --al-x=5.0000 --al-y=5.0000 --software=LinuxCNC
Importing preamble... DONE
Importing postamble... DONE
Importing front side... DONE.
Importing back side... DONE.
Importing outline... DONE.
Exporting back... 
Warning: pcb2gcode hasn't been able to fulfill all clearance requirements and tried a best effort approach instead. You may want to check the g-code output and possibly use a smaller milling width.
Makefile:2: recipe for target 'all' failed
make: *** [all] Segmentation fault
```
Howto debug this?

koppi, 2015-11-19:
Yay – I just played  with the above Makefile, changed it to ```--tile-x=2 --tile-y=2``` [...] ```--software=Custom```, this produces quite promising results:

* ```openscam ngc/arduino-shield_outline.ngc```:
[image]
* ```openscam ngc/arduino-shield_front.ngc```:
[image]
* ```openscam ngc/arduino-shield_drill.ngc``` – the drill file part gets somehow not correct:
[image]

corna, 2015-11-19:
The drill file is ok too. The weird path is caused by the optimisation: if, for example, you have two bit sizes, pcb2gcode drills the holes for the first bit on all the boards, then it changes bit and it does the remaining ones. With this workflow you have to change each bit only once.
Just to be sure, can you post the drill.ngc file?

I don't understand the segfault error though, I'll investigate.

koppi, 2015-11-19:
Here's the ```arduino-shield_drill.ngc```: http://paste.ubuntu.com/13350539/ When I open the file in OpenSCAM I can see, that the boards overlap vertically and there's too much space horizontally.

corna, 2015-11-19:
You're right, I didn't noticed the overlap; I'll investigate

corna, 2015-11-20:
It should work now, I've fixed the segfault and the overlap bugs in commits corna/pcb2gcode@250d7b9dceb96e90a18909557c4a4ade5f202c4f and corna/pcb2gcode@255378f1ecb9110fd9c5e4feeac0f42774879c7d.
Can you test it? Thanks.

koppi, 2015-11-20:
Thank's for the quick fixes.

One small thing I noticed, which can be seen in the screenshots above: the origin of the ```arduino-shield_drill.ngc``` file is different to the origin of the ```arduino-shield_front.ngc``` and ```arduino-shield_outline.ngc```.

corna, 2015-11-20:
That's because you've used ```--drill-front=false```, try with ```--drill-front=true```
(Side note: in you're repo there's no license file; did you do that on purpose?)

koppi, 2015-11-20:
Thank's for the hint with the drill file, that worked.

For the license file part, do you have any suggestions? I was using the KiCAD library at https://github.com/freetronics/freetronics_kicad_library – not being a lawyer, can you recommend what to do in this case?

corna, 2015-11-21:
[freetronics_kicad_library](https://github.com/freetronics/freetronics_kicad_library) is released under under CC-BY-NC (see [here](https://github.com/freetronics/freetronics_kicad_library/blob/master/LICENSE)), so you must release your project under either the same license (C BY-NC-SA) or under a [compatible license](https://creativecommons.org/compatiblelicenses). Unfortunately the author have added the NonCommercial clause, therefore the GPL is probably not compatible.
TL;DR: use the same license, [CC BY-NC-SA 4.0](https://creativecommons.org/licenses/by-nc-sa/4.0/)

closed 2015-11-21 by corna