pcb2gcode/pcb2gcode #17: Support bridge depth option

matthijskooijman, 2015-10-28:
I like the awesome new "bridges" options, they make my life so much easier (I used to modify the g-code by hand to do this). One thing I would love to see added, is to allow the bridges to be partially milled. Right now, the bridges are left completely intact which makes them hard to break out (I made them 3mm wide, perhaps smaller ones would be easier).

When doing this manually, I let the bridges be milled away about halfway through the PCB thickness, leaving enough material to keep the PCB in place, but allowing to easily break it away.

I believe this should be fairly easy to implement. I'm not sure if I'll find the time to actually dive into this anytime soon, but wanted to share the idea anyway :-)

corna, 2015-10-28:
Something like zbridges? ;-)

     --zbridges unit
     bridges heigth (Z-coordinates while engraving bridges, default to zsafe)

matthijskooijman, 2015-10-28:
Yeah, exactly that! I guess I missed it because it didn't start with "bridge" :-p

Thanks for your quick response, and also the great work on pcb2gcode in the last months. I haven't used it for a while and got out the latest version again to build a PCB today, and I could trash all of my workarounds, with everything working perfectly out of the box. Keep up the great work!

closed 2015-10-28 by matthijskooijman

corna, 2015-10-28:
No problem, glad to see you found pcb2gcode useful! :-)