pcb2gcode/pcb2gcode #22: Empty output depending on fill-outline=false

zazolabs, 2015-11-25:
Hi,

If i set  fill-outline to false outline.ngc is generated ok
but back/front are empty, i guess it have something to do with 

    Warning: pcb2gcode hasn't been able to fulfill all clearance requirements and tried a best effort     approach instead. You may want to check the g-code output and possibly use a smaller milling width.
    Can't create 4 bridges on this layer, only 3 will be created.

if set to true output is correct. /as excepted no internal cutoff-s are created/

[millproject.txt](https://github.com/pcb2gcode/pcb2gcode/files/44046/millproject.txt)

corna, 2015-11-25:
See issue #7 for the empty front/back, issue #18 for the internal cutoffs

corna, 2015-11-25:
Btw, I'm working right now on the fill-outline option, I'm going to enable it by default

zazolabs, 2015-11-25:
I'll take a deeper look at #7 and #18 but have to dig deeper to understand them.

closed 2015-11-27 by corna