pcb2gcode/pcb2gcode #23: drill from the back side is broken

zazolabs, 2015-11-25:
I've checked up the code and when drilling from the back side both mirror absolute and center are broken.

mirror absolute does -x but it should be board width - x, the problem is the board width must be the outline width and drill.cpp have/uses (?) the size of the drill part which is effectively smaller than the board and that results in an miss alignment. 

mirror along the center again must do real_board_width - current calculation.
If drill.cpp have the outline width it would be easy but failed to find it.

corna, 2015-11-27:
I've fixed some alignment bugs in commit corna/pcb2gcode@1c5b7c5f9c4a5af017138240e0d8d74116f08cce, pull the changes from [my repo](https://github.com/corna/pcb2gcode) and check if your problem is fixed

zazolabs, 2015-11-30:
Thank you, i'll run some tests and let you know .

EdgarJRobles, 2015-12-05:
[image]
The original file has ".drd" extension
Scale trouble and may offset too

corna, 2015-12-06:
It seems to me that your excellon drill file is broken, if you open it in gerbv/gerbview you can see that it is ~1m long
Maybe it is related to #4?
Which CAD are you using?

EdgarJRobles, 2015-12-06:
it come from Eagle inches to mm

matthijskooijman, 2015-12-16:
I just drilled a board from the back yesterday, using 4897d928dea3297bc8d50828fdef97a63863311f (plus the patch for #18), which worked exactly as expected. The board was generated using Kicad. So it seems this bug is specific to the workflow from Eagle?

corna, 2015-12-16:
This is the second time that Eagle generates an invalid gerber (see #4), so it is probably not pcb2gcode's fault.
Reopen if you think that the bug is on our side.

closed 2015-12-16 by corna

zazolabs, 2015-12-21:
A bit late but - tested and looks good.