pcb2gcode/pcb2gcode #25: Internal Error: Outside path reaches image margins at 36984,50)

matthijskooijman, 2015-12-06:
While playing with the testcase for #18, I got an internal error:

```
$ pcb2gcode --back noname-B.Cu.gbr --outline noname-Edge.Cuts.gbr
Importing front side... not specified.
Importing back side... DONE.
Importing outline... DONE.
Internal Error: Outside path reaches image margins at 36984,50)

Importing drill... not specified.
END.
```

The gerber files used can be found here: https://gist.github.com/matthijskooijman/bac820f2c5cae599413a They are generated from the testcase for #18, but the backside contains just a single square pad, without any ground plane or internal gaps, so I think this is a separate issue. The above error was with git master (without the internal gap changes added).

corna, 2015-12-16:
This bug is related to #18, keep the discussion on that issue.

matthijskooijman, 2015-12-16:
Are you sure? The error also occurs without the patch for #18 applied, and without any internal gaps in the gerber file?

corna, 2015-12-16:
Weird, I've never had this problem without the patch for #18, and I've noticed that this bug occurs when the background is not recognized as "background" during the internal cutoff process.
I'm reopening, I'll check it

matthijskooijman, 2015-12-16:
I just doublechecked using git master and the above files, and it indeed gives this error.

corna, 2015-12-16:
Ok, thanks

corna, 2016-04-21:
This is your gerber plotted (blue is back, red is edge cuts)

[image]

I suppose you're using fill-outline but, as you can see, the external border is not a closed line; I think you've inserted the mark in the top-left accidentally.

closed 2016-04-21 by corna

corna, 2016-04-21:
After removing that cross everything works (with millproject_example)

[image]

`````
pcb2gcode --back noname-B.Cu.gbr --outline noname-Edge.Cuts.gbr                                                                                           [16:48:00]
Importing front side... not specified.
Importing back side... DONE.
Importing outline... DONE.
Exporting back... DONE. (Height: 54.1016mm Width: 54.1016mm)
Exporting outline... DONE. (Height: 54.1016mm Width: 54.1016mm) The board should be cut from the BACK side. 
Importing drill... not specified.
END.
`````