pcb2gcode/pcb2gcode #26: z-bridges only applies on first pass

baldengineer, 2015-12-07:
Looks like z-bridge is only set during the first pass. On subsequent passes, it's ignored, cutting right through the bridge.

Picture/simulation for reference. Confirmed with real cut.

Attaching:
Reference Image:
[image]

Command Line Used (tried multiple z-bridge options: 0, positive, negative, all had same result):
[pcb2gcode-cmd.txt](https://github.com/pcb2gcode/pcb2gcode/files/53429/pcb2gcode-cmd.txt)

Generated gcode:
[outline.ngc.txt](https://github.com/pcb2gcode/pcb2gcode/files/53430/outline.ngc.txt)

Original Gerber:
[zigbeeshield-Edge.Cuts.gbr.txt](https://github.com/pcb2gcode/pcb2gcode/files/53432/zigbeeshield-Edge.Cuts.gbr.txt)

corna, 2015-12-08:
I've fixed it in commit 250d53ab9198f1f6f7f36b7e1deef5085d512693, can you test it?

baldengineer, 2015-12-09:
I only looked at simulated runs in Camotics, but looks like it is working as expected now. Thanks!

closed 2015-12-09 by baldengineer

corna, 2015-12-09:
You're welcome! Thanks for your bug report.