pcb2gcode/pcb2gcode #28: mill drill enchancemet

zazolabs, 2015-12-21:
It would be good to transform the mill-drill option to a second drill file.
As kicad can generate two drill files with plated/non-plated holes:
drill= as is 
milldrill= a drill file to be milled.

or other way

max-drill-size=1.5 -> all holes larger than 1.5mm use mill drill, otherwise use normal drill

hedefalk, 2018-01-31:
I would definitely like this too, the second option! I have a drl file with everything from 0.3mm to M4 holes and drill bits in the range 0.3mm - 1.2mm. 

If I understand the way it is now I can't mill with anything but the 0.3mm drill bit if I am to keep the 0.3mm holes? I mean I neither --onedrill or --mill-drill, helps me here? Maybe I'm missing an option?

So I would definitely like to set a limit like 1mm, anything bigger I can mill with the 1mm bit, but I wouldn't want to loose the change drill bit pauses.

I guess my workaround is to generate two gcode files from my one drl file, one with just

eyal0, 2018-06-13:
Is this still needed?  I know that it's been a long time.

If so, let me know.

hedefalk, 2018-06-14:
@eyal0 I think so! 

My rationale is in the related issue https://github.com/pcb2gcode/pcb2gcode/issues/45#issuecomment-361762545

It would be really nice to be flexible about the drill bits. I think a typical use case is that you have holes ranging from 0.2mm to say 4mm (mounting holes in my case) and maybe drill bits from 0.2mm to 1.2mm. (cheap kit from China).

So there one would really like to set a limit for the milling option so that everything under 1.2mm is drilled with the appropriate but but everything larger than 1.2 is milled with the largest bit.

If it's done by different files or gcode for tool change doesn't matter to me, but as I understand it there's no way to do this now and I've broken a few too small drill bits drying to mill with them :)

hedefalk, 2018-06-14:
Sorry, I hadn't kept up with the latest development. Maybe my use case is supported with the https://github.com/pcb2gcode/pcb2gcode/pull/147?

eyal0, 2018-06-14:
Okay, I can try to make it work.  Here are my questions:

Do you think that it's possible that someone will have more than 1 milling bit size?  I should think that a single milling bit would be enough because if it can mill a 2mm hole it can mill a 20mm hole, too.  The size of the milling bit doesn't effect the max size of the hole made, just the minimum.  So, can you think of a use case where more than 1 milling bit is needed?

Second, do you think that you'll always need everything *above* a certain size to be milled and everything *below* a certain size to be drilled?  Or is it possible that you want to drill 0.2 to 1.2, mill 1.2 to 3, and then drill for sizes 3-4.  My question: Is the milldrill limit just a minimum or do you think that it can be multiple ranges?

By knowing the answers so these questions, I can think about the best way to implement the options.

eyal0, 2018-06-14:
@hedefalk By the way, #147 doesn't exactly support your use case.  The latest code lets you specify the drill bits that you have and also the range for them.  Then when you are drilling, it will round to the nearest drill.  So if you say that you have 0.2,0.3,0.4,0.5 and you want to drill a hole that isn't exactly one of those, it will round to the nearest one.  This is useful if you have holes like 0.199 and 0.201 so that you don't have to keep "changing" drill bit when really those are the same number with a rounding error.

But this doesn't solve milldrill.  For drills that aren't rounded, they just get left alone.

hedefalk, 2018-06-14:
> Do you think that it's possible that someone will have more than 1 milling bit size? I should think that a single milling bit would be enough because if it can mill a 2mm hole it can mill a 20mm [...]

Possibly, but not me. I think my use case is simply to have a big enough mill bit that doesn't break and do everything above a certain diameter with that one, in my case a 1.2mm I think :)

> Second, do you think that you'll always need everything above a certain size to be milled and everything below a certain size to be drilled? Or is it possible that you want to drill 0.2 to 1.2, mill [...]

Again, I'm fine with just setting an upper limit and have anything bigger than that be drillmilled with my 1.2mm bit. Of course I could see the benefits of drilling 3mm holes with 3mm bits, but for me, that's not the culprit, I'm fine with just one limit.

zazolabs, 2018-06-14:
Yes, it is. One size is enough. if someone have a use case for more than one can tell us why?

PR #172 (2018-08-01): Add option to split drilling and milldrilling by hole diameter
Set the min-milldrill-hole-diameter to the minimum size hole that should be milldrilled.

This fixes #28 and #45
PR #176 (2018-08-01): Add zmilldrill option
This fixes #28 and fixes #45
closed 2018-08-01 by eyal0