pcb2gcode/pcb2gcode #31: Add alignment holes in pcb2gcode (new feature suggestion)

Hitman009, 2016-02-06:
For double sided PCBs, alignment holes are needed to flip the board when mirror-absolute=false. 

I wrote a command line tool in C# to add alignment holes (2 or 4). I think similar approach could be added into pcb2gcode

Input parameters: (can be improved)
AddAlignmentHoles = true/false
NumAlighnmentHoles = 2 or 4
AlignmentHolesYOffset = 0.8 (any numeric value)

For 2 alignment holes, 
I add 2 holes at 
(x=center, y=yMin-AlignmentHolesYOffset )
(x=center, y=yMax+AlignmentHolesYOffset )

For 4 alignment holes, 
I add 4 holes at
(x=center-xWidth/4, y=yMin-AlignmentHolesYOffset )
(x=center-xWidth/4, y=yMax+AlignmentHolesYOffset )
(x=center+xWidth/4, y=yMin-AlignmentHolesYOffset )
(x=center+xWidth/4, y=yMax+AlignmentHolesYOffset )

corna, 2016-09-28:
It can be done, but how should pcb2gcode behave if the PCB is completely occupied and there's no space left for the alignment holes? I think that they should be created by the PCB designer, not by the CAM processor

eyal0, 2018-04-02:
@Hitman009 Now that there is mirror-axis, it should be possible to reliably create alignment holes.  Just select a mirror-axis value is that near the middle of the x range of the CNC and then drill holes that are equidistance from the line X=mirror_axis.  Put pegs in those holes and line them up when flipping the board.

Like @corna said, boards might not have extra space around them so it's hard to come up with a generic solution.  For example, your solution assumes that the FR4 will have extra space in the Y axis but for some people, there will only be extra space in the X-axis.

Re-open this if you wan to continue discussing it!  Thanks.

closed 2018-04-02 by eyal0

c-, 2021-01-14:
> Like @corna said, boards might not have extra space around them so it's hard to come up with a generic solution.

I've been fighting with this issue... you need to put alignment holes in a location where they won't interfere with, say, auto levelling, but also where they'll stay within the bounds of the project.

The simple solution I've come up with (using a post-processing script) is to just place my alignment holes in the center of the outline cutter path (but not on the bridges, obviously). The outline is the last milling operation you'd normally do on a double-sided board and that makes it an idea sacrificial location for alignment holes. The only constraint is that the alignment drill is smaller than the outline cutter.