pcb2gcode/pcb2gcode #32: 1) Drill file does not start spindle (bug?), 2)IJ mode is not set to incremental (forgotten?)

Hitman009, 2016-02-08:
I am using pcb2gcode v1.2.1. (I think the situation is the same as the latest version v1.2.2). Those are the two problems i am facing: 

- Generated drill.ngc files do not contain a spindle start command (M3 ( Spindle on clockwise. )) although back, frond, and outline files contain this line. 

- It seems that pcb2gcode uses incremental IJ mode. If mach3 is set to IJ mode = absolute, gcode arc movements are totally different than the desired movement. I add G91.1 ( command = set IJ mode to incremental) in front of my ngc files. I think that G91.1 should be added to generated ngc files in pcb2gcode.

Hitman009, 2016-02-08:
For missing M3 command, here is my scenario: 

Here is the command line to run pcb2gcode
`pcb2gcode\pcb2gcode.exe --noconfigfile --drill output/alignment.drl --drill-output output/alignment.ngc --preamble preamble.ngc --postamble postamble.ngc --dpi 1000 --metric --metricoutput --zsafe 5 --zchange 20 --extra-passes 0 --mill-feed 60 --mill-speed 20000 --offset 0.2000 --zwork -0.1200 --nog81 --drill-feed 30 --drill-speed 20000 --milldrill --zdrill -5.0000 --cut-feed 60 --cut-infeed 1.0000 --cut-speed 20000 --cutter-diameter 0.1000 --fill-outline --outline-width 0.2000 --zbridges -0.6000 --zcut -5.000 `

I am using a preamble file containing (G91.1 and M3 commands). There is no M3 inserted by pcb2gcode. Here is the generated drill ngc file 

`( pcb2gcode 1.2.1 )

( This file uses a mill head of 0.10000mm to drill the 1bit sizes. )
( Bit sizes: [0.80000] )

G91.1
M3
G94       (Millimeters per minute feed rate.)
G21       (Units == Millimeters.)
G90       (Absolute coordinates.)
S20000    (RPM spindle speed.)
F60.00000 (Feedrate)

G00 Z5.00000
G0 X60.53180 Y-1.00000
G1 Z[-5.00000+4*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+3*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+2*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+1*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+0*1.00000]
G2 I-0.34380 J0
G0 Z5.00000

G0 X60.53180 Y74.25000
G1 Z[-5.00000+4*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+3*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+2*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+1*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+0*1.00000]
G2 I-0.34380 J0
G0 Z5.00000

G0 X20.40580 Y-1.00000
G1 Z[-5.00000+4*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+3*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+2*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+1*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+0*1.00000]
G2 I-0.34380 J0
G0 Z5.00000

G0 X20.40580 Y74.25000
G1 Z[-5.00000+4*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+3*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+2*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+1*1.00000]
G2 I-0.34380 J0
G1 Z[-5.00000+0*1.00000]
G2 I-0.34380 J0
G0 Z5.00000

G00 Z20.00000 ( All done -- retract )

M5      (Spindle off.)
M9      (Coolant off.)
M2      (Program end.)

`

corna, 2016-02-17:
M3 in the header was added in commit 3b2083254b5185f386e1dd7f2a046f9348ffe6fd
I've just committed the G91.1 change in commit e484b7098cfb5b5d9033ad2366d433ea842ff6f5

closed 2016-02-17 by corna