pcb2gcode/pcb2gcode #33: MacOS X: Compile time errors with Glibmm

danst0, 2016-04-17:
Hi,

I try to compile pcb2gcode (1.2.2) on my mac (el capitan). I came quite far, however now I found an obstacle:

`In file included from surface.cpp:23:
In file included from ./surface.hpp:34:
/usr/local/Cellar/glibmm/2.46.3/include/glibmm-2.4/glibmm/refptr.h:95:18: error: expected ';' at end of declaration list
  inline RefPtr() noexcept;
                 ^
                 ;`

I am not able to overcome it. Any hints?

Daniel

corna, 2016-04-19:
Hi,

recent version of glibmm require c++11, if you pull the latest commits the c++11 is enabled by default
Keep me updated, I'm interested in a Mac build

danst0, 2016-04-19:
Well. I got it working on my iMac. However now on my Macbook there is another error. Please try yourself.
`
$ brew tap danst0/pcb2gcode 
$ brew install pcb2gcode
`

corna, 2016-04-19:
Unfortunately I don't have Mac, I'll try to set up a VM

corna, 2016-04-25:
I've tried on a Yosemite VM and on a MacBook Pro and it works on both. From your formula I see that you're using version 1.2.2, but glibmm's fixes are after that release, so you have to pull from git.
I'm about to release a v1.3.0 (which will contain the glibmm's fixes), in the meanwhile you can use this formula
[pcb2gcode.rb](https://github.com/pcb2gcode/pcb2gcode/files/235265/pcb2gcode.txt)
install it with `brew install --HEAD`, or it will fail
Does it work?

danst0, 2016-04-26:
Works like a charm!!

corna, 2016-04-27:
Great! When release 1.3.0 I'll make a PR on the homebrew's repo

closed 2016-04-27 by corna