pcb2gcode/pcb2gcode #34: MSYS64: Compile errors "reference to ‘shared_ptr’ is ambiguous"

cryptomountain, 2016-04-21:
I'm trying to build a 64 bit version in MSYS64 on a Windows 7 platform and am getting "reference to ‘shared_ptr’ is ambiguous". The output is identical to this bug report: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=805220. I've attempted to comb through and define the namespace on the references, but I'm sure this is not the right way to solve this. I'm working on the master branch, hoping to write in a 'tinyg' compatible mode. Any suggestions?

cryptomountain, 2016-04-21:
Note: I've added in '-std=c++11' to the CXXFLAGS in the Makefile

corna, 2016-04-21:
Which version are you using?

cryptomountain, 2016-04-21:
I forked the pcb2gcode last night and am on the master branch. I installed and updated MSYS last night, Win7 64bit. 

One other note, during the prep of MSYS, I inadvertently installed the 32 bit versions of mingw-w64-i686-gcc, mingw-w64-i686-boost, and mingw-w64-i686-gtkmm, then installed the 64 bit versions and removed the 32 bit versions via pacman. ( not sure if that could be the problem or not).

corna, 2016-04-21:
Weird, I fixed this kind of bugs in e6580db4365c9f3ad42cea103382faeaf472800f
By the way, '-std=c++11' is useless, C++11 has been enabled by default in commit c4413c61115181faca8d5c548ea0b6f03e4140cc (as reported in bug #19)
I'll try with my Win10 VM

cryptomountain, 2016-04-21:
ok, that WAS really weird. I checked the 'configure.ac' file in my fork, and it was from Sept. 13, so it didn't have the fixed code. I deleted the forked repo in my acct, and re-forked. It built without any problem.
Issue closed, but no idea how that happened.

Thank you for helping through it, and great work on this project!

closed 2016-04-21 by cryptomountain

corna, 2016-04-21:
Great!
Thanks! :)