pcb2gcode/pcb2gcode #37: --vectorial outline

koppi, 2016-06-24:
```bash
$ pcb2gcode --basename=ngc/plot --drill=bumps-fab.drl --back=bumps-fab-5-CuBottom.gbl --front=bumps-fab-4-CuTop.gtl --outline=bumps-fab-0-EdgeCuts.gm1 --output-dir=. --dpi=600 --metric=true --metricoutput=true --mirror-absolute=false --optimise=true --zchange=40.0000 --zero-start=true --zsafe=0.5000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --zwork=-0.11 --drill-feed=1000 --drill-front=true --drill-speed=20000 --milldrill=true --nog81=false --onedrill=true --zdrill=-2.0 --cut-front=true --cut-feed=600 --cut-infeed=10.0000 --cut-speed=10000 --cutter-diameter=2.0000 --zcut=-1.6500 --al-probefeed=100 --al-x=5.0000 --al-y=5.0000 --software=custom --bridges=2 --bridgesnum=4 --vectorial --offset=100 --al-back=false --al-front=false
```

Is the code for the outline with the --vectorial option already in place? With the above options I see:

[image]

corna, 2016-06-25:
It should has been fixed in d3ee28dc68d41537de9268bf345b36800cb162bb, can you confirm it?

koppi, 2016-06-26:
I'm super excited about the voronoi feature. With your latest patch the outline works as expected, thank's a lot @corna .

closed 2016-06-26 by koppi

corna, 2016-06-26:
You're welcome