pcb2gcode/pcb2gcode #38: ExcellonProcessor::millhole function not used

koppi, 2016-06-24:
I would like to mill some holes, that are bigger than my milling tool. In pcb2gcode/drill.cpp I see the ExcellonProcessor::millhole function. But it is not used in the code base.

corna, 2016-06-26:
It is used in [drill.cpp:464](https://github.com/pcb2gcode/pcb2gcode/blob/2d3b1082bc589b03ddfe8b28b3ae9f9da3ff84b2/drill.cpp#L464) and activated by the command-line option _--milldrill_

For example:

     pcb2gcode --tile-x=2 --tile-y=2 --basename=ngc/arduino-shield --drill=arduino-shield.drl --front=arduino-shield-F_Cu.gbr --back=arduino-shield-B_Cu.gbr --outline=arduino-shield-Cmts_User.gbr --output-dir=. --postamble=ngc/postamble.ngc --preamble=ngc/preamble.ngc --dpi=1000 --metric=true --metricoutput=true --mirror-absolute=false --optimise=true --zchange=40.0000 --zero-start=true --zsafe=0.5000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=1.0 --zwork=-0.075 --drill-feed=1000 --drill-front=true --drill-speed=20000 --milldrill=false --nog81=false --onedrill=true --zdrill=-2.0 --cut-front=true --cut-feed=600 --cut-infeed=10.0000 --cut-speed=10000 --cutter-diameter=0.3000 --fill-outline=true --outline-width=1 --zcut=-1.6500 --al-back=false --al-front=false --al-probefeed=100 --al-x=10.0000 --al-y=10.0000 --software=LinuxCNC --milldrill

[image]

Unfortunately a combination of drill and mill hole is not supported yet, see #28

koppi, 2016-06-26:
I did a ```grep -R ExcellonProcessor::millhole```, not a ```grep -R millhole```. Thank's @corna for the clarification and the command-line invocation example.

closed 2016-06-26 by koppi