pcb2gcode/pcb2gcode #39: The tiling feature is broken

koppi, 2016-06-26:
See screenshot from my [arduino-shild](https://github.com/koppi/arduino-shield) repo:

[image]

Command-line was:
```bash
$ pcb2gcode --tile-x=2 --tile-y=2 --basename=ngc/arduino-shield --drill=arduino-shield.drl --front=arduino-shield-F_Cu.gbr --back=arduino-shield-B_Cu.gbr --outline=arduino-shield-Cmts_User.gbr --output-dir=. --postamble=ngc/postamble.ngc --preamble=ngc/preamble.ngc --dpi=1000 --metric=true --metricoutput=true --mirror-absolute=false --optimise=true --zchange=40.0000 --zero-start=true --zsafe=0.5000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=1.0 --zwork=-0.075 --drill-feed=1000 --drill-front=true --drill-speed=20000 --milldrill=false --nog81=false --onedrill=true --zdrill=-2.0 --cut-front=true --cut-feed=600 --cut-infeed=10.0000 --cut-speed=10000 --cutter-diameter=2.0000 --fill-outline=true --outline-width=1 --zcut=-1.6500 --al-back=false --al-front=false --al-probefeed=100 --al-x=10.0000 --al-y=10.0000 --software=LinuxCNC
```

corna, 2016-06-26:
I can't reproduce the bug with pcb2gcode 1.3.0 or HEAD and the command you posted, are you sure you're using the latest version?

[image]

koppi, 2016-06-26:
Seems to be only an issue in the development version of https://github.com/CauldronDevelopmentLLC/CAMotics then, closing.

closed 2016-06-26 by koppi