pcb2gcode/pcb2gcode #40: G91.1 in drill header prevents drilling

popsharma, 2016-07-01:
The line:
preamble += "G91.1     (Incremental arc distance mode.)\n";
in drill.cpp, causes the drill head on iModela machines to move off the work area in a diagonal fashion.
It is as though it is moving relative to the actual coordinates in the drill file.
Removing the G91.1 line from the final drill file output fixes the problem and drilling works as expected.

corna, 2016-07-02:
I've never had any problem with G91.1 and it should be correct (IJK values are indeed relative)
Can you please send me a screenshot of your output and a sample gerber + gcode?

popsharma, 2016-07-02:
Hi,
please find attached all of the files I could think of you needing.
does-not-work-drill.ngc           original output from pcb2gcode --nog81does-work-drill.ngc                  same output file with G91.1 line removed, works perfectlymill project                                settings file for pcb2gcode --nog81tutorial1.drl                               gerber output from KiCad 4.0.2
All other files work as expected.
I am using a Mac to run KiCad and pcb2gcode.I am using a PC to mill / drill the boards.
I am not sure what screenshot you were referring to, so I have not included any.
Hope this helps

Date: Sat, 2 Jul 2016 02:03:27 -0700

corna, 2016-07-02:
Unfortunately GitHub does not allow attachments in the email comments. Please save your file in a [gist](https://gist.github.com/) and post here the link

corna, 2016-07-02:
While reading the Roland iModela documentation ([here](http://icreate.rolanddg.com/iModela/download/dl/manual/NC_CODE_EN.pdf) and [here](http://www.rolanddg.co.uk/files/iM-01_Master_Guide_R1.pdf)) I discovered that Roland does not support G91.1. I suppose that your software interprets G91.1 (incremental **arc** distance mode) as G91 (incremental distance mode)
Can you swap the G90 and G91.1 lines and try if it works (**without** removing G91.1)?

popsharma, 2016-07-02:
Hi,
please see below the link I got from the GIST page:
<script src="https://gist.github.com/popsharma/aa622d2d850ecfae3dd9b1e33bd73643.js"></script>

Date: Sat, 2 Jul 2016 03:41:31 -0700

corna, 2016-07-03:
I can correctly simulate both your files with LinuxCNC, it should be a Roland's bug
I've added a __--nog91-1__ switch in commit bcb1a9eba12c234e0a176373882afac90b9385e0 to remove the G91.1 line from the header

closed 2016-07-03 by corna