pcb2gcode/pcb2gcode #43: Add graphical documentation

chrysn, 2016-07-04:
Some time back I've added a visualization of the various pcb2gcode options, but missed to send them upstream. They can be seen in my graphical-documentation branch at https://github.com/chrysn-pull-requests/pcb2gcode/blob/graphical-documentation/man/options.svg .

I haven't looked over the options added or changes since then; would you consider such a graphic practical for documentation? If you have options you think that could still be included, or you think something is misrepresented, I'd be happy to update it.

corna, 2016-07-05:
I like it, but:
 * _--zdrill_, _--zwork_, _--zcut_, _--zchange_ and _--zsafe_ should be represented as "ranges" (like _--cut-infeed_)
 * _milldrilling_ should be replaced with _drilling/milldrilling_
 * _--back_ should be replaced with _--front/--back_
 * we should add _--cut-feed_
 * maybe we should add another block explaining _--bridges_ and _--zbridges_

chrysn, 2016-07-06:
I've done some updates, though not all you suggested:

* The Z are now ranges (except for zafe, where things would be too crowded and there's no obvious visual reference plane anyway; the faded range should work for people who see the rest)
* milldrilling is changed
* w/rt back: I don't think that the image could take illustrations for working with two settings. Isn't --back the common case for single-setting setups?
* cut-feed: Changed. Please check whether they now refer to the right speeds, back then both steps (engraving and outline) took the same --mill-feed afair.
* bridges: won't go well with this illustration I think, plus I'm not really familiar with them, could do another one though. Will a --zbridges=-0.5mm --cut-infeed=0.6mm mean that the first pass of outline cutting will go all the way around, and only the last two will spare the bridges?

chrysn, 2016-07-06:
(Something went wrong in the image, there's a blue bar now displaying, can dig into that later).

corna, 2016-07-07:
_--cut-feed_ and _--zwork_ are barely visible, the rest is ok

> * w/rt back: I don't think that the image could take illustrations for working with two settings. Isn't --back the common case for single-setting setups?

It depends, for example in my SMD circuits i always use _--front_ only (as both the tracks and the components are on the same layer). I think that replacing _--back_ with _--front/--back_ is ok (everything is in place both for _--front_ or _--back_)

> * bridges: won't go well with this illustration I think, plus I'm not really familiar with them, could do another one though. Will a --zbridges=-0.5mm --cut-infeed=0.6mm mean that the first pass of outline cutting will go all the way around, and only the last two will spare the bridges?

With _--zbridges=-0.5_ and _--cut-infeed=0.6_ all the steps will spare the bridges, with the opposite (_--zbridges=-0.6_ and _--cut-infeed=0.5_) yes.

Here you can see the effect of _zcut=-2.5_, _cut-infeed=0.25_ and _zbridges=-0.6_

[image]

Btw I've fixed a couple of bridges and multi-step cutting in the commits 29b5a5ef977f614b7b5a2242d50067ea3422d975 and 69b0e64d0f32be60462ab0fb74b6a3f4b37ed617, pull them

corna, 2016-07-16:
I'm about to release a v1.3.2 and I want to include this svg, can you fix the image?

chrysn, 2016-07-17:
probably not during this week, sorry, i'm at a conference and (judging from the last two days) busy 19h/day. sorry, this will need to wait for the next one.

corna, 2016-09-07:
Hey @chrysn, news about this issue?

chrysn, 2016-09-07:
i've just pushed an update to the branch:

* the editing artifacts are gone now (not sure what got messed up inside inkscape, but a partial revert fixed it); those had caused the `--zwork` visibility issues
* `--cut-feed` visibility got enhanced

corna, 2016-09-14:
Great, thanks
Which license do you want to use for this image? CC-BY?

chrysn, 2016-09-14:
the image says cc-by-sa in its metadata, that's what i had intended
originally.

but to make things simpler: you, as the maintainer of pcb2gcode, may
publish any contribution i made or will make to pcb2gcode under any
license you deem suitable.

corna, 2016-09-14:
I forgot the SA part, CC-BY-SA is ok for me (I didn't know there is a "license" field inside SVGs)

corna, 2016-09-28:
Added with ac820935fd2654d32e3919f67f58cd29cbf5cab8, in the next release I'll integrate it better in the CLI doc and inside the GUI
Thanks!

closed 2016-09-28 by corna