pcb2gcode/pcb2gcode #44: Error after building from source

TadGhostal, 2016-07-11:
I have compiled the current git version on Mint 17 with boost 1.61 and everything seems to build fine. When I run the program I get this error:

Error: You've supplied an invalid parameter.
Note that spindle speeds are integers!
Details: character conversion failed

I get this error when I try ./pcb2gcode --help I still get the same error. The much older apt repository version seems to work fine but it looks like the command line arguments may not with the pcb2gcodeGUI. Is there a fix for this?

corna, 2016-07-11:
I'm not having any problem with the git head (built on Debian testing with Boost 1.58).
Are you sure there isn't any configuration file in the current directory? Try to run it with _--noconfigfile_
I'm experiencing multiple issues with Boost 1.61, probably generated by some bugs in the library, I suggest you to use another version (Boost 1.60 works well). If your distribution provides only Boost 1.61 you can download the [sources of another version](http://www.boost.org/users/history/), build them, and compile pcb2gcode (autoreconf -fvi && ./configure --with-boost=\<boost_dir\> --enable-static-boost && make)

TadGhostal, 2016-07-12:
I built  and linked 1.60 and that seemed to fix the problem.

closed 2016-07-12 by TadGhostal

corna, 2016-07-12:
I've just realized that it shouldn't work on 1.60 (and, on my PC, it doesn't work), see [#11905](https://svn.boost.org/trac/boost/ticket/11905)
Have you manually built Boost 1.60 from upstream?

TadGhostal, 2016-07-13:
I am not sure, I got it from here: https://sourceforge.net/projects/boost/files/boost/1.60.0/ then built and installed that version. All I know is that after I did this and rebuilt pcb2gcode it worked in the terminal and so did pcb2gcodeGUI.