pcb2gcode/pcb2gcode #45: Enhancement - skip small holes when using --milldrill

popsharma, 2016-07-31:
Is it possible to add a feature to skip drilling holes smaller than the endmill cutter diameter when using --milldrill? 
These smaller holes could be drilled in a separate pass using a normal drill file.
e.g. --milldrill-skip-small

Now only holes >= cutter diameter will be drilled.
The user could do a second pass with a 'normal' drill file to drill the smaller holes with a drill bit. No special feature required for this.

corna, 2016-09-28:
Not so easy for how the ExcellonProcessor class is made, but I'm planning to add in the next release _milldrill_behaviour_ that accepts 3 values:
 * _skip_, to skip all the holes smaller than _milldrill_diameter_
 * _ignore_, to mill them anyways
 * _drill_, to mill the holes larger and drill the smaller ones

This would merge this issue with #28

popsharma, 2016-11-01:
That's a great idea. It covers all the bases I think.

Looking forward to it.

hedefalk, 2018-01-30:
I would definitely like this too I have a drl file with everything from 0.3mm to M4 holes and drill bits in the range 0.3mm - 1.2mm. 

If I understand the way it is now I can't mill with anything but the 0.3mm drill bit if I am to keep the 0.3mm holes that small? I mean I neither --onedrill or --mill-drill, helps me here? Maybe I'm missing an option?

So I would definitely like to set a limit like 1mm, anything bigger I can mill with the 1mm bit, but I wouldn't want to loose the change drill bit pauses.

I guess my workaround is to generate two gcode files from my one drl file, one normally with just drills and then a second one using milldrill where I remove all the small holes by hand/script… Or is there an easier workaround?

PR #176 (2018-08-01): Add zmilldrill option
This fixes #28 and fixes #45
closed 2018-08-01 by eyal0