pcb2gcode/pcb2gcode #49: compile from source results in boost error

mattvenn, 2016-09-06:
Following the instructions carefully:

checking for Boost headers version >= 1.56.0... /usr/local/lib/boost_1_57_0
checking for Boost's header version... 1_57
checking for the toolset name used by Boost for g++... gcc48 -gcc
checking boost/program_options.hpp usability... yes
checking boost/program_options.hpp presence... yes
checking for boost/program_options.hpp... yes
checking for the Boost program_options library... no
configure: error: cannot find the flags to link with Boost program_options

ubuntu 14.04

Linux mattsmac 3.13.0-53-generic #89-Ubuntu SMP Wed May 20 10:34:39 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

corna, 2016-09-06:
Try installing the boost libraries somewhere with

     ./bootstrap.sh --with-libraries=program_options --prefix=<somewhere>
     ./b2 variant=release link=static
     ./b2 install

Does it work?

mattvenn, 2016-09-07:
yes that worked. Do you want me to do a pull request to change the docs or
will you do it?

corna, 2016-09-07:
Already done in 064c91b29b5ace167641c4502fd0965c6df14b77, thank you

closed 2016-09-07 by corna