pcb2gcode/pcb2gcode #51: Spindle spin up time

snajpa, 2016-09-29:
Hi,

let me first thank you for this awesome tool, great job!

At the moment, we have to do some post-processing to files generated via pcb2gcode, so I thought I could suggest the improvements to you, as maybe more people could find them useful (so I'll be filing maybe a few more tickets after this one).

First and perhaps the most important is this one:

Our spindle takes about 10 seconds to fully start up to 24k RPM.

It would be really nice to have an option to add a dwell period, that would ensure that the spindle is fast enought before cutting into the material. I think it could be a common value for both mill and drill steps.

Sample gcode:

```
S24000
M03
G04 P10.000000 (Dwell 10s after spindle start)
(Do something useful here)
M05
M02
```

If you find this interesting enough to implement, please let me know via mail where can I send you some beer in return (snajpa at snajpa.net).

/snajpa

corna, 2016-09-30:
Glad to know you found it useful!

I've added _--spinup-time_ and _--spindown-time_ in commit corna/pcb2gcode@1c6f6d75d79d74cbbbf9bf674e88c520ed73f2ca, can you test it?
Feel free to open more tickets for different requests.

I've also added a **donate** button in the README.md if you want to buy me a beer :)

snajpa, 2016-09-30:
Thanks a lot!

I'm going to test this today.

Btw, could you please add this to pcb2gcodeGUI as well? Should I create a ticket there too?

I can't find the donate button :(

corna, 2016-09-30:
No need to create a ticket there, as soon as you confirm me that it works I add it to pcb2gcodeGUI

I've just added it, you can find it in the README.md ([this](https://www.paypal.com/cgi-bin/webscr?cmd=_s-xclick&hosted_button_id=KDT9RVZ9Y2NA4) is the link btw)

corna, 2016-09-30:
Are you using pcb2gcodeGUI? On which platform?

snajpa, 2016-09-30:
Can you please send the PP link in english? :)

We're using pcb2gcode with pcb2gcodeGUI on Fedora (currently 23) here in our hackerspace in Brno, CZ (feel free to stop by if you're ever in the area!).

Here's a repo that I've cobbled together with your two tools: http://repo.vpsfree.cz/fedora-eda/

Our plan is to include these two (and some more) tools in Fedora 25 at the earliest, though 26 seems more realistic.

A bit more on our setup, if you find it interesting - the machine is a chinese cnc-based construction with ball screws (google CNC 3020Z). Control is based on a Core i7 x86 machine running LinuxCNC. We're currently working on integrating your tools with our tool, which does the probing and g-code transformation according to it - https://github.com/hackerspace/hacked_cnc. The goal is that you'd provide all the gerber (and excellon) files, choose where to place them and it would just let you go trought the process with a simple wizard, so you'd only need to do the tool changing. It's a work in progress, currently it only does the levelling (but using bi-linear transformation, it does it really well).
As far as the parameters of the machine go - the lowest resolution it can do is 0.03mm, it can go about 500mm/min; we're using chinese carbid cnc tools with a diameter of 0.1mm (these don't last that much) and 0.2mm (much better at the longevity).

Also, feel free to come by and have a chat with us on our IRC channel (#base48 at freenode).

corna, 2016-09-30:
(Modified the language of the donate button, thank you)

Cool, we have exactly the same CNC machine :)
Btw pcb2gcode already have a levelling feature (see the _autoleveller_ section in the manual) that embeds the probing and bilinear transformation directly in the gcode, check it out

Tomorrow I'll come on IRC for sure, thank you :)

snajpa, 2016-09-30:
We use contact probe, the PCB is pulled up to 5V via a resistor, then the tool tip grounds it via the machine. So we have to pre-probe the whole grid and then transform the gcode.

The command we're using is 'G38.1 Z-100 F30', which means go down in Z until you touch the PCB or hit -100mm, speed 30mm/min.

Is that going to work with what pcb2gcode exports?

I'm currently having trouble exporting anything at all with 1.3.2 (and your branch too), trying to figure out what is going on.

corna, 2016-09-30:
I have the same setup (PCB pulled at 5V, tool at GND) and the exported gcode is similar (G38.2 Z-3 F80), but the advantage with pcb2gcode's autoleveller is that you don't have to probe the PCB and then modify the gcode in a separate step.

As an example look at this file: [front.ngc](https://gist.github.com/corna/2ba8e840951d3820a9b8323a66efe13a)
As you can see at the beginning the gcode executes a grid probe and save the Z values in the gcode variables. Then, instead of using the standard G01 for the movements, it uses a "o2" subroutine, that is a G01 with a Z correction based on the current "probe rectangle" and the previous probed values.
In this way the gcode dynamically adapts to the current PCB surface.

But, if you don't like this autoleveller, you can still disable it and get a plain gcode (that you can easily modify).

What's the problem with the exporting?

snajpa, 2016-10-01:
Turns out I've mixed up the versions, so that confused me a bit.

Anyway, the previous version we have seems to work just fine (it's custom build just before 1.2.2 version tag), however, current git master doesn't. It outputs just about a shell of the G-code it is actually supposed to spit out - that's with the new vectorial core on, fall back on the older one fails altogether.

The files are a bit cluttered with the leveling code, as that's a neat thing I'm playing around with now, so I'm sorry for that. But the leveling is uber cool! Didn't know you could do it like this... (though doing the transformation separately from probing has its own advantages, such as being able to level all the input files, ie. if you don't drill that deep).

here are the input files:

espregulator.front.gbr:
https://paste.vpsfree.cz/9p1_T54p/
espregulator.drill.drd:
https://paste.vpsfree.cz/iWpvkBIC/
espregulator.outline.gbr:
https://paste.vpsfree.cz/zEt9d99s/

and here are the outputs in vectorial mode:

command line:
```
pcb2gcode --drill=/home/snajpa/Documents/eagle/espregulator/espregulator.drill.drd --front=/home/snajpa/Documents/eagle/espregulator/espregulator.front.gbr --outline=/home/snajpa/Documents/eagle/espregulator/espregulator.outline.gbr --output-dir=/home/snajpa/Documents/eagle/espregulator/export-new2 --dpi=1000 --metric=true --metricoutput=true --mirror-absolute=false --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=false --zchange=10.0000 --zero-start=true --zsafe=2.0000 --extra-passes=6 --mill-feed=600 --mill-speed=24000 --offset=0.1500 --zwork=-0.0400 --drill-feed=100 --drill-side=auto --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=true --zdrill=-1.8000 --bridges=0.5000 --bridgesnum=2 --cut-feed=400 --cut-infeed=1.8000 --cut-side=auto --cut-speed=10000 --cutter-diameter=3.1750 --fill-outline=false --zbridges=-0.6000 --zcut=-1.8000 --al-back=false --al-front=true --al-probefeed=100 --al-x=10.0000 --al-y=10.0000 --software=LinuxCNC
```
output
```
Importing front side... DONE.
Importing back side... not specified.
Importing outline... DONE.
Processing input files... DONE.
Exporting front... DONE. (Height: 46.4466mm Width: 92.3546mm)
Exporting outline... DONE. (Height: 46.4466mm Width: 92.3546mm) The board should be cut from the FRONT side. 
Importing drill... DONE.
Exporting drill... DONE. The board should be drilled from the FRONT side.
END.
```
front:
https://paste.vpsfree.cz/47NIaUoW/
drill:
https://paste.vpsfree.cz/9zWQTMQ0/
outline:
https://paste.vpsfree.cz/E2o4JNTC/

And here's the output in the old mode:

command line:
```
pcb2gcode --drill=/home/snajpa/Documents/eagle/espregulator/espregulator.drill.drd --front=/home/snajpa/Documents/eagle/espregulator/espregulator.front.gbr --outline=/home/snajpa/Documents/eagle/espregulator/espregulator.outline.gbr --output-dir=/home/snajpa/Documents/eagle/espregulator/export-new2 --metric=true --metricoutput=true --mirror-absolute=false --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=10.0000 --zero-start=true --zsafe=2.0000 --extra-passes=6 --mill-feed=600 --mill-speed=24000 --offset=0.1500 --voronoi=false --zwork=-0.0400 --drill-feed=100 --drill-side=auto --drill-speed=10000 --milldrill=false --nog81=false --nog91-1=false --onedrill=true --zdrill=-1.8000 --bridges=0.5000 --bridgesnum=2 --cut-feed=400 --cut-infeed=1.8000 --cut-side=auto --cut-speed=10000 --cutter-diameter=3.1750 --fill-outline=false --zbridges=-0.6000 --zcut=-1.8000 --al-back=false --al-front=true --al-probefeed=100 --al-x=10.0000 --al-y=10.0000 --software=LinuxCNC
```
output:
```
Importing front side... DONE.
Importing back side... not specified.
Importing outline... DONE.
Processing input files... DONE.
Exporting front... Internal Error: Failed repairing @ (1208,1380)

Importing drill... DONE.
Exporting drill... DONE. The board should be drilled from the FRONT side.
END.
```
drill:
https://paste.vpsfree.cz/x-35GUr9/

corna, 2016-10-01:
I can't open your files (page not found), can you check the links?

What do you mean with

> such as being able to level all the input files, ie. if you don't drill that deep

corna, 2016-10-05:
(moved the commit here corna/pcb2gcode@1c6f6d75d79d74cbbbf9bf674e88c520ed73f2ca)

snajpa, 2016-10-05:
I'm super busy right now, I'll get to this no sooner than ~Tuesday next week. The issue seems to be only with the vectorial core and gerber files from Eagle, though I still need to give it a few more tries to be sure.

corna, 2016-11-13:
Have you tested it?

eyal0, 2018-03-29:
I see spinup and spindown added in https://github.com/pcb2gcode/pcb2gcode/commit/3b2d8f8ed52b12719005edd11adb2c8c05cb1069 so I'm closing this issue.  Re-open if needed.

closed 2018-03-29 by eyal0