pcb2gcode/pcb2gcode #52: drill from the back side is broken (like #23) but with Vectorial core

goranskular, 2016-10-12:
With the Vectorial core back side mirroring is not working properly and it is not aligned with the drill. After a lot of tweaking, I tried with disabled victorial core and then it works properly.

It is probably the same bug in calculation.

corna, 2016-10-13:
Can you post here a sample Gerber with the commands used, so that I can reproduce the error?
Which version of pcb2gcode are you using?

goranskular, 2016-10-13:
It works with:

pcb2gcode.exe --back=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/antenna_analyser-B.Cu.gbr --drill=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/antenna_analyser.drl --outline=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/antenna_analyser-Edge.Cuts.gbr --output-dir=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/mach3 --dpi=1000 --metric=true --metricoutput=true --mirror-absolute=false --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=false --zchange=40.0000 --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --zwork=-0.0500 --drill-feed=1000 --drill-side=auto --drill-speed=10000 --milldrill=true --milldrill-diameter=1.0000 --nog81=false --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=600 --cut-infeed=10.0000 --cut-side=auto --cut-speed=10000 --cutter-diameter=1.0000 --fill-outline=true --outline-width=0.2000 --zbridges=-0.6000 --zcut=-2.5000

It does not work with:

pcb2gcode.exe --back=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/antenna_analyser-B.Cu.gbr --drill=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/antenna_analyser.drl --outline=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/antenna_analyser-Edge.Cuts.gbr --output-dir=C:/Users/goran/Downloads/antenna-analyser-goran/antenna-analyser-master/cnc/mach3 --metric=true --metricoutput=true --mirror-absolute=false --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=40.0000 --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --voronoi=false --zwork=-0.0500 --drill-feed=1000 --drill-side=auto --drill-speed=10000 --milldrill=true --milldrill-diameter=1.0000 --nog81=false --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=600 --cut-infeed=10.0000 --cut-side=auto --cut-speed=10000 --cutter-diameter=1.0000 --fill-outline=true --outline-width=0.2000 --zbridges=-0.6000 --zcut=-2.5000

corna, 2016-10-13:
Are you using the git head or the latest release (v1.3.2)?

goranskular, 2016-10-13:
yes. 1.3.2 and gui 1.3.2.1

corna, 2016-10-13:
I think that the fix for this bug is in commit e19b1d2eadd2740fb356e08f3693cfaa0439d25e, committed after v1.3.2
Compile the latest git version (or, if you're using Windows, download [this binary](
https://sourceforge.net/projects/pcb2gcode/files/pcb2gcode/test%20versions/pcb2gcode-1.3.2-git9798728-win32.zip/download) and unzip it in the pcb2gcodeGUI folder); is the bug fixed now?

goranskular, 2016-10-13:
Yes. Now it is working correctly. Thanks.

closed 2016-10-13 by corna

goranskular, 2016-10-13:
I found [pcb-probe](https://github.com/atom-bomb/pcb-probe) 

It seems it can do a job.