pcb2gcode/pcb2gcode #53: Pre drill during etch

fcobcn, 2016-10-26:
Would it be possible to add a pre drill pass to the etch file?
Something like what eagle pcbgcode does, this helps to avoid the drill from slipping on the pcb.
I will be happy to buy the developer some beers for this functionality.

corna, 2016-11-03:
It's not easy to add, but it is a useful functionality, I'll try to implement it
Beers are always appreciated :)

eyal0, 2017-12-17:
@fcobcn What if you just set a shallow drilling depth and then run the drill.ngc file?  That would pre drill the holes, right?

fcobcn, 2017-12-17:
@eyal that’s a good workaround. Thanks for the idea

eyal0, 2018-02-19:
@fcobcn Is that workaround enough?  Can we close this issue?

fcobcn, 2018-02-20:
@eyal0 Haven't tried yet, but should work OK.

closed 2018-02-20 by fcobcn