pcb2gcode/pcb2gcode #55: Outline broken for vectorial core in 9e9fe6f

eyal0, 2016-11-10:
The fill-outline was broken by the commit 9e9fe6f .  Please read my last comment here:

https://github.com/pcb2gcode/pcb2gcode/issues/7

I checked with e19b1d2 and the code works in that version from 5 weeks ago.  Latest code makes an empty back.ngc.

eyal0, 2016-11-11:
This is an ugly but working solution:

https://github.com/eyal0/pcb2gcode/commit/93f456494933330355e0cb59bb463ee9ecd6dba0

It might not solve all cases.  The problem is that the layer, after parsed, might have paths that are not contiguous.  For example, if the edge cut is a square, it might arrive as one path for one side and another path with three sides.  Also, they might link front to back or front to front.

My overly simple solution was to loop through all the paths and look for any path that can be connected to another path.  In all those cases, I connected them and deleted the unused one.

Because the edge cuts are a list of undirected edges, essentially, the best solution might be to create an entire graph of all the edges and then look for all cycles (rings) in those edges.  If edges cross each other it'll be tricky but you get the idea.

corna, 2016-11-11:
I took a similar approach in corna@c11923f
I think that this solution is enough, creating a graph seems to me overcomplicated
Can you test it?

eyal0, 2016-11-11:
I agree, no need to overcomplicate. I will add tests. I hope to have it
ready in 24 hours.

Eyal

eyal0, 2016-11-11:
I've added the tests and made a pull request for you.  I was able to see that without simplify_paths, the output was incorrect the n, and with, it was correct.  I hope that you can merge it successfully.

corna, 2016-11-13:
Great, thanks for the testing
Merged (c11923f29bb038665451debb63d79fbc6593318f)

closed 2016-11-13 by corna