pcb2gcode/pcb2gcode #57: Quick move to zsafe after switching tools.

eyal0, 2016-12-07:
I made a drill file that uses two different sized drills. The generated gcode has tool changes and pauses in it, which is great. However, after the tool change, the CNC makes a quick lateral move to the first drill hole X,Y and then plunges into the material from the tool change height. This happens at feed rate so it's very slow.

The rest of the holes work as expected, quick moves laterally at zsafe height and slow Z moves between zsafe and zdrill.

I think that it would be enough to either add G0 Z(zsafe) for the first move after tool change or just add the Z(zsafe) to the first G0 after the tool change.

Let me know if you need further assistance or examples. Or if you need me to make the patch.

eyal0, 2016-12-08:
Here are the first lines of a sample drill file:

( Software-independent Gcode )

( This file uses only one drill bit. Forced by 'onedrill' option )

G94       (Millimeters per minute feed rate.)
G21       (Units == Millimeters.)
G90       (Absolute coordinates.)
S10000     (RPM spindle speed.)

G00 Z50.00000 (Retract)
T1
M5      (Spindle stop.)
(MSG, Change tool bit to drill size 0.80000 mm)
M6      (Tool change.)
M0      (Temporary machine stop.)
M3      (Spindle on clockwise.)

F30.00000
G0 X6.70580 Y18.13580
G1 Z-1.60000
G1 Z5.00000
G0 X10.51580 Y20.67580
G1 Z-1.60000
G1 Z5.00000

Most G1Z commands that do the drilling will start from zsafe, 5, but the first one will start from zchange, which is 50!  That's a very long time to plunge down to -1.6.

corna, 2016-12-09:
Done in ac263e4a993711bb8cb13056e780fcad341d6792, tell me if it works

eyal0, 2017-05-17:
Works

closed 2017-05-17 by eyal0