pcb2gcode/pcb2gcode #58: Build fails: libgerbv.so: undefined reference to `screen'

phidaisllc, 2017-01-02:
I read an issue from 2014 very similar to this, but no clear solution:

make  all-recursive
make[1]: Entering directory '/home/jerry/src/pcb2gcode'
Making all in man
make[2]: Entering directory '/home/jerry/src/pcb2gcode/man'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/jerry/src/pcb2gcode/man'
make[2]: Entering directory '/home/jerry/src/pcb2gcode'
  CXX      autoleveller.o
  CXX      board.o
  CXX      common.o
  CXX      core.o
  CXX      drill.o
  CXX      gerberimporter.o
  CXX      layer.o
  CXX      ngc_exporter.o
  CXX      surface.o
  CXX      surface_vectorial.o
  CXX      tile.o
  CXX      options.o
  CXX      outline_bridges.o
  CXX      voronoi.o
  CXX      main.o
  CXXLD    pcb2gcode
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/libgerbv.so: undefined reference to `screen'
collect2: error: ld returned 1 exit status
Makefile:561: recipe for target 'pcb2gcode' failed
make[2]: *** [pcb2gcode] Error 1
make[2]: Leaving directory '/home/jerry/src/pcb2gcode'
Makefile:623: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/jerry/src/pcb2gcode'
Makefile:458: recipe for target 'all' failed
make: *** [all] Error 2

phragment, 2017-01-03:
This problem occurs with libgerbv version 2.6.1, either use 2.6.0 or a later vcs version. I can provide further help, if you provide more info about your system.

phidaisllc, 2017-01-12:
Thank you very much. A downgrade solved the problem.

closed 2017-01-12 by phidaisllc

corna, 2017-01-28:
The issue was caused by a bug introduced by gerbv 2.6.1, later fixed in commit dd8f9fa
I've opened an issue on Debian and the fix has been cherrypicked in testing/sid

lhartmann, 2018-07-26:
As of Fedora 28 the issue is still on the distro. A simple and ugly workaround (no package manager involved) is:

- Download older gerbv from [Fedora 27 repository](http://fedora.c3sl.ufpr.br/linux/releases/27/Everything/x86_64/os/Packages/g/gerbv-2.6.0-11.fc27.x86_64.rpm).
- Extract to a temporary folder with 7z (twice) or use a gui (ark) to browse the archive contents.
- Find and TEMP/usr/lib64/libgerbv.so.1.0.7 and copy it to /usr/lib64.
- Point existing symlinks to it:
```
cd /usr/lib64
sudo ln -sf libgerbv.so.1.0.7 libgerbv.so.1
sudo ln -sf libgerbv.so.1.0.7 libgerbv.so
```
Should work fine, and the original file is still there as libgerbv.so.1.0.8 if anything goes wrong.

cacheflood, 2018-11-08:
Still issue on Fedora 29.
(4.18.16-300.fc29.x86_64 #1 SMP Sat Oct 20 23:24:08 UTC 2018 GNU/Linux)