pcb2gcode/pcb2gcode #59: Exporting front... Segmentation fault: 11

staviq, 2017-01-11:
When using --vectorial --voronoi, it either complains about intersecting geometry which is not true, or, compiled from head ( brew install pcb2gcode --HEAD ) gives me segfault:

output: http://pastebin.com/YRPHk411

input files below:

millproject: http://pastebin.com/P5ttDJEj

2seg.GTL: http://pastebin.com/0W1Xbm8d

2seg.GML: http://pastebin.com/gg9VTu9k

source eagle files:

2seg.brd: http://pastebin.com/jW4BTE3S

cam job for eagle: http://pastebin.com/rfHz4HEL

eyal0, 2018-02-05:
I started work on this and I was able to reproduce the bug.  It seems that, when generating voronoi regions, it skipped some.  Those skipped regions are null shapes and that led to the segfault.

I tested with both the older voronoi building code from about 2 months ago and the newer one and both of them show the same behavior.  I was also able to distill 2seg.GTL down to just 4 shapes and still reproduce the bug.

I'll work on this.

eyal0, 2018-02-06:
Here's a file with a smaller test case that fails:

[2seg.tar.gz](https://github.com/pcb2gcode/pcb2gcode/files/1698605/2seg.tar.gz)

Just run pcb2gcode in that directory.

I've determined that the problem is `bg::intersection` on integral types.  Converting to floating point before running `bg::intersection` and then converting back solves the problem.

closed 2018-02-07 by eyal0