pcb2gcode/pcb2gcode #60: Most paths get milled twice using voronoi mode

zagl, 2017-01-14:
With the current git version (2017-01-12) when using voronoi mode every every copper piece is milled completely. Thus, all borders between touching surfaces get milled twice.

For example I used this Gerber layout:
[image]

This is traced as follows in voronoi mode:
[image]

In the following picture I marked every single mill path. It is obvious that nearly every path is milled twice.
[image]

One advantage of using voronoi could be that it is possible to nearly half the processing time. But in this case the processing time is nearly the same or even longer.

Is it possible to detect already milled paths and skip these?

eyal0, 2017-12-17:
@zagl I'm working on this.  If I get it working, would you be interested in testing it?

eyal0, 2017-12-17:
Here's a pretty good attempt:

[image]

We can see in the image that the two paths for the input are visually identical.  However, the left path is about half the length because it doesn't do each trace twice, only once.

eyal0, 2018-02-01:
This issue is solved with eulerian-paths mode in vectorial, which is on by default.

closed 2018-02-01 by eyal0