pcb2gcode/pcb2gcode #61: Inverted gerbers

smooker, 2017-01-18:
I got:
Processing input files... Non-positive image polarity is deprecated by the Gerber standard and unsupported; re-run pcb2gcode without the --vectorial flag
terminate called after throwing an instance of 'gerber_exception'
  what():  std::exception

However without the vectorial core - it produces quite the same output (like positive gerber) ;)
This way it's impossible to make an solder mask gcode for laser engraving.

corna, 2017-04-03:
That error means that your Gerber is using the polarity setting in a way which has been deprecated by the current Gerber standard, and the vectorial core supports only the latest Gerber specification revision.
Maybe it can be implemented, can you upload a sample file?

eyal0, 2018-03-28:
Too long without a response.  Please re-open if this is still an issue.

closed 2018-03-28 by eyal0