pcb2gcode/pcb2gcode #62: Windows build fails with boost 1.63

kosme, 2017-01-25:
Currently, the build of pcb2gcode fails on Windows when ./configure is run on a system where boost 1.63 is installed. The message error is something like: 

`checking for boost/geometry.hpp... no`
`configure: error: cannot find boost/geometry.hpp`

The workaround for this is to downgrade to boost 1.62 by using the following command for x86_64:

`pacman -U http://repo.msys2.org/mingw/x86_64/mingw-w64-x86_64-boost-1.62.0-1-any.pkg.tar.xz`

or this one for i686:

`pacman -U http://repo.msys2.org/mingw/x86_64/mingw-w64-i686-boost-1.62.0-1-any.pkg.tar.xz`

Then ./configure will exit gracefully.

eyal0, 2018-02-19:
@kosme Sounds like you've already found the solution.  Are you able to test on newer boost versions? 
 Do they work?

Can we close this issue?

kosme, 2018-02-23:
It is working with boost 1.66

closed 2018-02-23 by kosme