pcb2gcode/pcb2gcode #63: Traces get milled twice

siredmar, 2017-03-24:
The issue occurs when traces are sorrounded by e.g. a ground polygon with a given isolate width.
This example shows the issue with a very small toolset offset 0.05mm (tool has ~0.1mm diameter).

See attached image: 
Left: PCB design
Middle: Gerbers and output from pcb2gcodeGUI
Right: Output gcode
[image]

The GCode could be optimized that only traces are being milled but not the the ground plate sorrounding the traces. Pins get isolated from ground plate the first time anyway.
Of course within the ground plate the thermals and pin connections would be necessary.

Is it possible to add a switch like: Detect already milled paths (within a given range) and don't mill them again.

eyal0, 2017-12-12:
@siredmar I'm working on this.  If I get something working, are you interested in testing it?

siredmar, 2017-12-17:
@eyal0: yes, i would definitely test it!

eyal0, 2017-12-17:
@siredmar By the way, have you tried using voronoi=true?  Maybe that will solve the problem for you well enough?

siredmar, 2017-12-18:
@eyal0: tried that, but didn't came out very well...

eyal0, 2017-12-19:
@siredmar Okay.  There are a few ways to tackle this.  For sake of discussion, I'll look at a little section of the multivibrator example:

[image]

Here's what it looks like if I set the offset larger:

[image]

The part where the green and white traces are separate get traced twice, because they are far apart, but the part where the green and white traces have bumped into each other, the milling overlaps.  In that section, the milling currently happens twice, also.  I could fix that so that the overlapping parts happen just once.

However, the *whole* trace got wider, not just the part where they were close.  Is that a problem?  Can you try increasing the offset and see if you like the results?  Don't worry if the overlapped parts get milled twice, I'll solve that soon enough.

eyal0, 2018-02-19:
Is this still an issue?  Why did you not like the results with `voronoi=true`?

eyal0, 2018-03-28:
There's not much to do about this for traces that are simply near but not exactly.  Using eulerian paths (which is default on) and setting the tool width correctly and also possibly using voronoi mode should give good results.

There's no way for pcb2gcode to know when you want to just drill a single path or when to drill two of them so there's no consistent solution to this problem in my opinion.

closed 2018-03-28 by eyal0