pcb2gcode/pcb2gcode #64: Milled drill centres offset in X from where they should be

deirdreobyrne, 2017-05-06:
When setting pcb2gcode to mill my holes with a 0.2mm milling bit, I get the attached. As you can see all the holes are offset, but only in the x-axis.
[image]
I'm using the Windows GUI, and a typical command looks like this -
pcb2gcode.exe--back="..." --drill="..." --outline="..." --output-dir="..."  --metric=true --metricoutput=true --mirror-absolute=false --nog64=false --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=10.0000 --zero-start=true --zsafe=5.0000 --extra-passes=0 --mill-feed=200 --mill-speed=10000 --offset=0.1000 --zwork=-0.2000 --drill-feed=200 --drill-side=auto --drill-speed=10000 --milldrill=true --milldrill-diameter=0.2000 --nog81=false --nog91-1=false --onedrill=false --zdrill=-0.5000 --bridges=0.5000 --bridgesnum=2 --cut-feed=200 --cut-infeed=10.0000 --cut-side=auto --cut-speed=10000 --cutter-diameter=0.2000 --fill-outline=true --outline-width=0.2000 --zbridges=-0.6000 --zcut=-0.5000

eyal0, 2017-05-16:
I'm working on the milldrill code in my fork and I think that I might solve this problem as part of my work.  Can you provide some more information?

Can you attach or paste the drill file?  Specifically, does it have any G85 codes in it?  I want to see if the holes are simply in the wrong place or if they are also the wrong size.  If you can, also attach the same photo but with milldrill=false.

Thanks!

deirdreobyrne, 2017-05-17:
Thanks for the response. The "good" news is that I got it to work once, the issue then became intermittent, and I'm now unable to reproduce it! It *may* have had something to do with my KiCad settings. If I see the issue again I'll be sure to save the gerber files so that you can reproduce it at will.

May I take this opportunity to put in a request? Would it be possible to mill pilot holes instead of milling the whole hole? i.e. drop whatever tool I have on the machine to the depth specified in the command line without trying to mill the hole outline, and without changing tools? The reason I ask is because I have a cheap Chinese CNC machine, and I haven't found a way of drilling right through without either damaging the CNC bed, or risk not having my PCB parallel to the machine axes. If I could mill pilot holes it would make manual drilling a lot easier, and an awful lot more accurate!

Thanks.

deirdreobyrne, 2017-05-17:
... I now seem to have managed to reproduce the issue. Try the following gerber and drill files (which were generated by TopoR).

[test.zip](https://github.com/pcb2gcode/pcb2gcode/files/1006290/test.zip)

eyal0, 2017-05-17:
Please send drill.ngc, too.  Thanks!

deirdreobyrne, 2017-05-17:
I get the same result as you when using gerbv. Changing that line in path.drl fixes what the design looks like in gerbv, but it does nothing to the output of pcb2gcode - the drill holes are still quite far from where they should be!

I'm using pcb2gcode on windows version 1.3.2

I also use camotics to preview what the gcode will produce on a CNC (http://camotics.org/download.html). It shows that the drill holes will be milled quite far from where they should be. And I've past experience of camotics which shows that when it says drill holes won't be milled in the right place, then that's what actually happens on the CNC.

eyal0, 2017-05-17:
Hmm.  Even after the repair suggested above, I see that gerbv is now working for me but pcb2gcode is showing me an offset in the drill holes, like you're seeing.  Oddly, the offset that I'm seeing is much larger than yours!

deirdreobyrne, 2017-05-17:
Yes - this example has a much larger offset than the example I gave above. I think we are both seeing exactly the same thing!

eyal0, 2017-05-17:
I'm using bCNC and see the offset, too.  I think that there are two issues here:
1. gerbv misinterpreting the tool size.  That's fixed as I recommend above.
2. pcb2gcode milling off center.  That's a pcb2gcode bug, not gerbv it seems, because gerbv is rendering it okay.

eyal0, 2017-05-18:
Eureka!

Change `--back` to `--front` to solve your problem.

I don't have your original files but you should make sure that the mask that you are looking at is on the correct side.  Did you intend to have a board with front copper only?

deirdreobyrne, 2017-05-19:
I'm afraid that isn't going to work as a solution.

I certainly intended to produce a single-sided PCB. I usually only buy single-sided PCBs - instead of aligning two sides of a PCB during milling, I just try and route everything on one side (the back!) and use the occasional wire on the front where I cannot route. I find it works well - I usually only need a few wires.

So when I do what you suggest, and use --front instead of --back, whereas the drill holes now align, the entire board is backwards from what I need!

BTW what do you think of the suggestion to add a feature to mill pilot holes instead of the entire hole?

eyal0, 2017-05-19:
Are you sure that, on the tool that you are using, you only used the back layer? Double check by comparing the Gerber file and the design.

I also do one sided boards and for me, everything aligns. I drill from the back. I use KiCad.

If your want to mill pilot holes, just turn off mill drill.

deirdreobyrne, 2017-05-21:
My flow is this -

- Design the circuit in kicad
- Layout a double-sided PCB, with most tracks on the back, and only put tracks on the front when I have to (those tracks will be wire links in the final product)

I've checked that everything aligns up correctly in gerbview.

So I do create a "double-sided" PCB, but I only use one side.

I've attached a set of files containing both layers and the drill file.
[files.zip](https://github.com/pcb2gcode/pcb2gcode/files/1017424/files.zip)

d-digiacomo, 2018-01-28:
I noticed holes were slightly offset along X with my latest PCB also. This is the first time it's happened - I fabricated a number of other designs a few months ago which worked well. Gerbers/.ngc files were generated through KiCad as outlined at https://www.baldengineer.com/kicad-to-x-carve-pcb-workflow.html. I'll be designing a few more PCBs later this week and will post on how they go.

[image]

eyal0, 2018-04-02:
@deirdreobyrne I did a lot of work with the double-sided code and there were some bugs in it.  I hope that I solved your issue along the way.  If you have the time, please check.  Try working with the `mirror-axis` and `zero-start` options to get the results you expect.  Good luck!

eyal0, 2018-04-29:
Haven't heard back in a while.  I'm closing this.  Reopen if needed.

SmallestLeslie, 2018-07-30:
Hi,
I'm totally new to GitHub so please excuse me if I  do something wrong.
And sorry for my English.
I having a similar issue with the offset drill holes.
Although I think not the drill holes are offset on the pcb rather the back is offset from everything else.
If I disable the vectorial core option the back and the holes are in the right place.
I use the pcb2gcode 1.3.2 and pcb2gcodeGUI 1.3.2-1 on W10 pro.
The gerbers are made with KiCad v4.0.7.
[image]
[image]
[image]

eyal0, 2018-07-31:
Can you provide the command line that you used to run this?  I need the configuration.

SmallestLeslie, 2018-07-31:
I did not save the configuration but I tried to reproduce the settings as far as I could.
"Luckily" these produce the same result.
[settings.txt](https://github.com/pcb2gcode/pcb2gcode/files/2244268/settings.txt)

eyal0, 2018-07-31:
[image]

For me it looks fine.  I used the settings that you send and the gerbers, too. You can see that the hole in the bottom left is aligned and also there is a tiny dot in the middle of each pad where the drilling would occur.

Can you check that you are using the latest pcb2gcode?  Run pcb2gcode --version and send the results.  Thanks!

eyal0, 2018-07-31:
I have never used the releases and they have not been updated in a long time.  Actually, I have no idea how to build the Windows versions, that's why I have never done it.  I also don't know how to use homebrew.  I don't know how to use msys2, either.  I'm sorry that I'm not very helpful!

I also don't use pcb2gcodeGUI because I use the command-line but I am considering trying to update it to support the latest features of the latest pcb2gcode.  I could actually use some help with that stuff.  If you want to take on the job of building a release or updating any of the pcb2gcodeGUI code, I could use the help!

SmallestLeslie, 2018-07-31:
When I wrote homebrew I was referring to the pcb2gcode instance I built myself.
It was misleading, sorry.
I made the builds based on the readmes posted on
https://github.com/pcb2gcode/pcb2gcodeGUI and
https://github.com/pcb2gcode/pcb2gcode.

I would like to help but I doubt I can.
I am an electric engineer and I did a few small projects with PIC uC programmed in assembly.
I did also one very small project in C# but it basically was googling and ctrl-c ctrl-v from stackoverflow. Generally I'm not familiar with high level languages.
Although it may be possible that I could do something useful after some learning.
And if you are interested I may have some suggestions for improvements.

For applications like this I definitely like to use GUI rather than command-line (I use linux with some struggle but I'm rather a windows type person).

eyal0, 2018-07-31:
> What I built from git does not allow to use board center to backside mirroring BUT the output it generates does NOT have the offset

I'm glad that the offset went away.  Could you try using the `mirror-axis` and `zero-start` options to see if you get results that you like better?

I added to the wiki a page about this topic: https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Two-sided-PCB-and-offsets

I hope that helps.  Let me know!

SmallestLeslie, 2018-07-31:
Do I understand right that if I want the centerline of the board as mirroring axis I need to manually calculate and give as argument to `--mirror-axis` the half of the width of the PCB?
Could not this be done by pcb2gcode?
Do pcb2gcode provide gcode for drilling the alignment pins?
I'm sorry if I went off topic here. Is there a forum or a platform where this could be talked through?

SmallestLeslie, 2018-07-31:
Ok, I got it. Then I write down what I think just as if I was thinking out loud. Please feel free to use it if it does make any sense.
On second thought this is indeed a problem difficult to solve universally.
The best method would be if the user would place two holes to the PCB no matter inside or outside the outline in the PCB design software and then the connecting line of this two holes could be defined as the mirroring axis.
This would be simple to specify but not with options rather by graphically.
Or even with options if the holes could be selected from rolldown lists from the drl file. This could be further simplified by filtering the holes by diameter.

Lets assume I would put two holes on the PCB with an unusual diameter. These two holes could be easily found by filtering to their diameter. Then it doesn't matter what size we drill them to (practically to the diameter of the available alignment pins).
Although I don't know how much energy would this take to realize it in program code.

Other: I would like to use the `autolevelling` option to compensate the unevenness of the PCB but pcb2gcode generate the probing raster offset from the board. Could you please check it?

Other2: If I enable the option `Optimise toolpaths` and the option `Extra passes` greater than 0 pcb2gcode fails to finish generation of toolpaths (for the back side). Enabling `autolevelling` makes it worse.
I think I can live without optimised toolpaths but I definitely would like to use autolevelling.
[files.zip](https://github.com/pcb2gcode/pcb2gcode/files/2246679/files.zip)

eyal0, 2018-08-01:
I tried enabling both optimise and extra-passes=2 and it worked fine for me.  Do you mean that it's taking a long time or does it never succeed?

I've never used autolevelling so I don't have a way to know if it's working!  I'm not sure how to help with that.  How can I fix it?

SmallestLeslie, 2018-08-01:
Is it possible that this is caused because of the 2.6.2 ver. of gerbv I installed?

eyal0, 2018-08-01:
Optimise isn't working for you because optimise is broken in boost. I didn't break it! https://svn.boost.org/trac10/ticket/13645

Switch to boost 1.66

As for autolevel, I can try to debug it but my gcode viewer doesn't support the autolevelling code. I use bCNC.

Can you tell me what you use to see the gcode like in your images above? Then I can use it, too, and I'll be able to debug the autolevelling for everyone!

SmallestLeslie, 2018-08-01:
I use camotics. It throws an error during procesding ngs files but it shows the toolpaths.

eyal0, 2018-08-01:
I'm unable to get camotics to work but I think that I understand the autolevelling code and I have a fix ready.  I would appreciate it if you could look at the results and let me know if the pattern of the levelling looks good.  I'm interested to know if it is accurately covering the milling area.  I also want to know if it works for front milling and for back milling.  Finally, I want to know if it works correctly with the options tile-x and tile-y.  If you could do those, too, that would be excellent!

eyal0, 2018-08-01:
I'll let you know when it's merged into master  For now, you can test it now with this branch if you like:

https://github.com/eyal0/pcb2gcode/tree/offset_autoleveller

eyal0, 2018-08-01:
@SmallestLeslie Okay, it's in now.  Let me know if you get around to trying it.

The issue on github has gotten really long and side-tracked.  Let's finish the autolevelling offset bug discussion here and if something new comes up, open a new issue.  That way it's easier to track what still needs fixing.

SmallestLeslie, 2018-08-02:
> Switch to boost 1.66

Unfortunately I haven't been able to get boost 1.66 work. It's definitely my fault. I wish I knew linux better. I built them with 1.67 and generated NGCs without optimisation.

> I would appreciate it if you could look at the results and let me know if the pattern of the levelling looks good.

Two sided milling with autolevelling (--tile-x=1 --tile-y=1) seems to work right now.
[image]

> I want to know if it works correctly with the options tile-x and tile-y. If you could do those, too, that would be excellent!

I tried a few variations with tile-x and tile-y:
--tile-x=2  --tile-y=1
[image]

--tile-x=1  --tile-y=2
[image]

--tile-x=2  --tile-y=2
[image]

SmallestLeslie, 2018-08-02:
Can I request to align the origin of the coordinate system to the bottom left corner of the pcb?

Can I ask for your opinion about my concept of defining the mirroring axis for double side milling I wrote above?

eyal0, 2018-08-03:
@SmallestLeslie That tiling and autolevlling looks really wrong.  I'm going to have to take a look at it.  I opened a new issue for it.  Thanks!

SmallestLeslie, 2018-08-03:
I did a milling today and the holes are located properly. Thank you for the fix.

Finally I was able to get boost 1.66 work so theoretically optimisation should work also.
However I do not see any difference with optimisation disabled or enabled. What should it do exactly?

I encountered a strange thing. The prebuilt release version generates gcode where the isolation areas are milled from the conductor outwards in a continous cut (in case of extra passes).
The lastly built version however generates gcode irregularly once inwards then outwards then to both directions. Once it cuts the isolations with climb method, another time with conventional method.
Do you experience this phenomenon?
[image]

eyal0, 2018-08-04:
You can open issues on GitHub, that is the best way to make requests.

Optimise option makes the gcode shorter by removing moves that can be removed and still maintain good fidelity. It's good that you don't see a difference. You can search wikipedia for Ramer Douglas Puecker algorithm or see this drawing:

https://camo.githubusercontent.com/534a4b86edff28e0f9c3123debf21c2bbdb91f4a/68747470733a2f2f7261772e6769746875622e636f6d2f77696b692f4c756b61737a57696b746f722f7365726965732d726564756365722f6578616d706c65732f6578616d706c65312e706e67

If you are unhappy with some paths drilled inward and some outward, you can turn off tsp_2opt and maybe that will help? pcb2gcode is trying to find the path that minimizes the number of times that the tool has to plunge and also minimize distances between paths. Sometimes it is faster to do a path in reverse so it does that. Turn off tsp_2opt and it won't reverse the paths.

Does it matter to you in which direction paths are done? Is it better to cut clockwise or counterclockwise around a trace? I am not a CNC expert. If there is a best direction, I could use that.

eyal0, 2018-08-04:
From my reading, it seems that climb milling gives the best finish. So if the tool spin clockwise when viewed from above, we want to go around the trace in a clockwise direction.

I'm not sure if that is what happens already. I'll check!

SmallestLeslie, 2018-08-04:
> I assume that to mill a hole we want the reverse direction.

I would like to point out that I have not milled holes yet. I drilled them all. So I don't have experience about hole milling but I assume that the same applies to it than trace milling.

> Which pass should be milled first, inside or outside?

Ok, I wasn't sure so I did another round of milling and I am a little bit confused now but I clearly concluded the following:
A continuous climb cut from inside (cutting counter-clockwise) outwards  results clean trace and burry rest-material (or how should we call it?).
A continuous climb cut from outside (cutting clockwise) inwards result burry trace and clean rest-material.

I think that our best bet (in case when at least 1 extra pass is set) would be to start milling with climb method from inside outwards until the penultimate pass then finish the cut with a conventional cut by reversing the direction of the cut.
Something like this:
[image]

eyal0, 2018-08-04:
Wow!  I really like your drawing.  Also, you've done great research.

This seems like it would be a useful feature because the burrs can make it difficult to later paint the board with curable paint (the green kind that you see on board).  I did it once and the paint can gather near the burrs.

I will try to add a feature.  I'll leave it up to the user to decide if climb-then-conventional or conventional-then-climb.  This feature may interfere with Eulerian paths and the path reversal, both of which are done to find shorter paths, so I'll make it so that when users enable the climb/conventional options, it disables eulerian and tsp_2opt reversals.  I'll assume that milldrilled holes work the same as milled traces and have them all done as if they are the last pass (so in your drawing, conventional).

Thanks!

SmallestLeslie, 2018-08-04:
I forgot to attach the difference:
[image]
I just figured out that unfortunately my 1/8" collet has about 0,07 mm runout which is in case of a 0,1 mm engraving tool, no good.
It is also depends on the machine and the feeds and speeds used but I doubt that a milling process could be done with no burr at all. A slight deburring always might be necessary.
Maybe I try to find the best position of the tool and try to make another try.

eyal0, 2018-08-08:
I did it!  Want to try it out?  Just add `--mill-feed-direction=climb` or `--mill-feed-direction=conventional`.

SmallestLeslie, 2018-08-08:
Did I something wrong?
At first pcb2gcode told me:
`Error: Can't use tsp-2opt together with mill-feed-direction`
So I added `--tsp-2opt=false`
With this it reach to importing outline, there it waits a little then quits.
```
Importing front side... not specified.
Importing back side... DONE.
Importing outline...
c:\P2Gcode>
```

SmallestLeslie, 2018-08-08:
OK, I don't know what I did, but now it works.

oleg-milantiev, 2018-11-20:
Hi.
First of all, thanks for great product!

Have drill shift too.

Files to reproduce bug: http://download.milantiev.com/pcb2gcode/
Problem exists only if choose outline. Command line (from windows GUI):

pcb2gcode.exe--back="myFocuserPro-B.Cu.gbr" --drill="myFocuserPro.drl" --outline="myFocuserPro-Margin.gbr" --output-dir="cut"  --metric=true --metricoutput=true --mirror-absolute=false --nog64=true --optimise=true --tile-x=1 --tile-y=1 --tolerance=0.0100 --vectorial=true --zchange=40.0000 --zero-start=true --zsafe=3.0000 --extra-passes=0 --mill-feed=450 --mill-speed=24000 --offset=0.1000 --zwork=-0.1000 --drill-feed=450 --drill-side=back --drill-speed=24000 --milldrill=false --nog81=true --nog91-1=false --onedrill=false --zdrill=-2.5000 --bridges=1.0000 --bridgesnum=4 --cut-feed=300 --cut-infeed=1.5000 --cut-side=back --cut-speed=18000 --cutter-diameter=3.0000 --fill-outline=true --outline-width=0.2000 --zbridges=-0.6000 --zcut=-2.0000

[image]

[image]

---

Strange thing. Its my old first KiCad project. Maybe I miss something about year ago? :)
Cause yesterday I draw new project (my second KiCad project) - it drill ok, outline superb and all work correct! Return to old project, regenerage gcode and remember about drill shift.

oleg-milantiev, 2018-11-24:
I'm using pcb2gcode-1.3.2-win64
but -V didt work :(

C:\PORTABLE\pcb2gcode-1.3.2-win64>pcb2gcode.exe -V
Error: You've supplied an invalid parameter.
Note that spindle speeds are integers!
Details: unrecognised option '-V'

eyal0, 2018-11-24:
Yes, that is an old version of the code and I have never compiled a new one... because I don't know how.  Maybe I should figure that out!

oleg-milantiev, 2018-11-24:
Did u mean no way to use new code at windows? Is latest version available on linux for example? I have one near...

Sorry, my english not so good, maybe i missunderstood your answer.

eyal0, 2018-11-24:
The releases only has a very old version of pcb2gcode.  If you compile from the source, you will get a much better version.  I know how to compile from source for linux but I don't know how to do it for Windows.

eyal0, 2019-02-23:
No progress here in a long time.  I'll close this.  Reopen a new bug if there are new issues.  Thanks!

closed 2019-02-23 by eyal0