pcb2gcode/pcb2gcode #68: Vectorial core crashes with some gerbers

christoph1703, 2017-05-25:
Firstly, a big thank you for this great piece of software. I've been using it successfully for some time now and never had problems until today.

I have a file which consistently causes the program to crash:
[esp-programmer-F.Cu.gbr.zip](https://github.com/pcb2gcode/pcb2gcode/files/1029434/esp-programmer-F.Cu.gbr.zip)

I have tried exporting the gerber file from KiCAD with different settings, no success. Other boards still work fine. It also works when I disable the vectorial core.
I'm using v1.3.2 on Linux Mint 18.

This is the output I get every time:
```
$ pcb2gcode --front=esp-programmer-F.Cu.gbr --output-dir=/ --vectorial=true --zchange=12.0000 --zchange-absolute=true --zero-start=true --zsafe=5.0000 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --zwork=-0.0500
Importing front side... DONE.
Importing back side... not specified.
Importing outline... not specified.
Processing input files... terminate called after throwing an instance of 'boost::geometry::overlay_invalid_input_exception'
  what():  Boost.Geometry Overlay invalid input exception
Aborted
```

Best regards
Christoph

eyal0, 2018-02-05:
I've been unable to reproduce this.  I tried using a few different versions of pcb2gcode and none of them crash.

Are you still seeing this bug?

christoph1703, 2018-02-11:
Just checked on my running system and in a freshly installed VM, and the error is still there with this one gerber file. I haven't had the error with any other file since then.

eyal0, 2018-02-11:
Weird.  Maybe it's due to the version of boost in the VM?  The error is from boost, not from pcb2gcode, and I'm not seeing it with my version of boost.

What version of boost are you using?  If you don't know, you can probably find it in the output of the `./configure` command.

christoph1703, 2018-02-11:
I'm using boost version 1.58, as recommended in readme. Installed it from the repos.

eyal0, 2018-02-11:
The code is automatically tested against 1.56 and 1.66.  Maybe 1.58 doesn't
work?  It's happenned that bosot broke and then broke something.

eyal0, 2018-05-16:
Okay!  I finally got around to figuring this out.  I noticed it while working on https://github.com/pcb2gcode/pcb2gcode/pull/132 .

Old versions of boost are broken.  The automatic testing doesn't catch it because only some inputs fail and the failing inputs aren't testing against the vectorial core.

I was able to reproduce the same bug as you with boost 1.56 but with boost 1.66, the problem goes away.  Other versions of boost might work, too, but I didn't test them all.

Are you able to use a newer version of boost?

christoph1703, 2018-05-21:
Just found the time to try boost 1.67 and it works! Thank you very much for your help!

closed 2018-05-21 by christoph1703