pcb2gcode/pcb2gcode #72: Enhancement: Laser mode

MrSurly, 2017-09-20:
Maybe allow generating G-Code for lasers:

* No Z movements
* Set spindle speed to 0 during rapid moves

ellisgl, 2017-10-14:
That would be an awesome feature.

eyal0, 2017-12-10:
Looks like you're not the only one working with lasers: https://github.com/pcb2gcode/pcb2gcode/pull/71

Can a laser be used to make a PCB?  How?

MrSurly, 2017-12-10:
@eyal0 : https://youtu.be/1hFNj86L7sk?t=6m36s

eyal0, 2017-12-10:
I don't know anything about gcode for lasers. I guess that z axis is unused? Is spindle speed used to control the laser power?

MrSurly, 2017-12-10:
> I don't know anything about gcode for lasers. I guess that z axis is unused? Is spindle speed used to control the laser power?

Yes, exactly.  Also, since the machine has motion planning, the speed of the XY movement isn't constant ; there's acceleration, etc. Some g-code interpreters (e.g. GRBL) have a "laser mode," which modulate the laser power based on the actual speed of movement, so that the laser doesn't burn more when the head slows down.  It will actually turn off the laser when the motion stops.

eyal0, 2017-12-10:
So, what are all the changes that need to be made to the output to make it work for laser?  [This PR](https://github.com/pcb2gcode/pcb2gcode/pull/71/files) seems to allow the user to do something before and after each milled section, maybe that would work for turning the laser off before rapid moves and on after rapid moves?

Also, I read about a controller that will do all G0 (rapid) moves with the laser off and turn it back on for G01, G02, G03.  So would that already turn off the laser as needed?

What happens if the laser CNC gets Z axis commands?  I would think that it would just ignore them, which would work just fine, right?  So you could just leave them in and it wouldn't do anything?

MrSurly, 2017-12-10:
@eyal0 
The specific problem I had is that pcb2gcode was generating z axis moves with the spindle on.  GRBL (which is my laser controller) does _not_ ignore Z moves, but since there's no Z axis, the laser pauses in one spot while the Z axis "moves".  This causes "hotspots".  I would manually remove any Z moves from the output g-code.  However, after finding that GRBL has a "laser mode," I don't really need to do any of that anymore.

MrSurly, 2018-05-06:
Sorry, missed the earlier email.  

The laser still fires on Z-axis movements, which it (arguably) shouldn't, though maybe there's a use case for that in GRBL laser mode.

eyal0, 2018-05-13:
I don't have a laser for my printer so I don't know what the ramifications are of firing during Z-axis movement.  Does this mean that the laser is running while the printer doing the Z move even though there is no z movement?  Does this cause the laser to shine on a spot for too long?  Because if there is no z axis, wouldn't the z-movement happen instantly, doing nothing, and then not affect the results?

MrSurly, 2018-05-13:
My laser runs stock GRBL, and laser mode only affects a couple of things.
It doesn't modify the Z axis in any way. So, it sends the pulses, laser
fires, and burns a spot. It's not instant, GRBL doesn't know/care about the
missing Z.

eyal0, 2018-05-17:
If you set the z values like `z-safe` and `z-cut` all to the same value, would that solve the problem?  Then grbl should be able to fly right over the z commands, right?

MrSurly, 2018-05-21:
>If you set the z values like z-safe and z-cut all to the same value, would that solve the problem? Then grbl should be able to fly right over the z commands, right?

```
./pcb2gcode --mill-speed=200 --mill-feed=500 --offset=0 --zwork=0 --zchange=0 --zsafe=0 --front <gerber file> 

Error: The safety height --zsafe is lower than the milling height --zwork. Are you sure this is correct?
```

Didn't see any way to assert that it was indeed correct.  Also, `--zsafe` is not lower than `--zwork`, so the error message is incorrect.

eyal0, 2018-05-26:
I see.  That's written as an error and the program exits when that happens.  It would be easy to modify the code to fix that.

For your situation, you've got a machine that has an x and y axis in use and the z axis isn't in use.  But the machine doesn't know that so when it sends Z axis commands, it sends power to the motor for the motor to do its thing and that takes time, even though the motor isn't connected and there's really nothing happening.  But it's wasting time and potentially the laser is on and lasing the surface too much.

Solutions that I see:

1. I fix up those error messages and then you set all the z values to the same value.  When the printer gets z-axis commands, since the z axis is already in the right place, it'll do nothing and so it will waste almost no time on those commands.  The only time wasted would be the time for the software to receive the z command and do nothing with it, which should be essentially zero.
2. I add an option to disable output of all z commands.  That's more complicated to implement but solves the problem.  The issue then is what if you *want* some z commands, just not all of them.  That's a problem.
3. Do your own post processing and remove all z commands.  Just search for Z followed by and number and remove those.  Not too hard but you might get errors.

I think that option 1 is best.  It's easy to do and it lets you still use z commands for some things, for example, maybe you do want a zsafe that is different from the z height for lasering.  I'll reword those error messages and then make them not cause pcb2gcode to exit.

a7hybnj2, 2018-08-09:
I just came across this issue and wanted to say: "YES!" it would be great to have a flag "--ignore-z-errors" or something similar. I think since there are a lot of millers out there the error message is a good thing so don't just get rid of it.

eyal0, 2018-10-02:
Oh!  I added a way to ignore z errors (and all other warnings).  Check out #201 and re-open this if you need more help.

hustav, 2023-07-12:
When zsafe and zwork are both zero, the gcode file does not match the silkscreen. The output is correct if I change both to another value, say 0.1.

For example, with zsafe = zwork = 0:
G00 X2.09064 Y34.89830 ( rapid move to begin. )
G01 F100.00000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z0.00000 ( retract )

Same code section generated with zsafe = zwork = 0.1:
G00 X2.09064 Y34.89830 ( rapid move to begin. )
G01 F100.00000
( Mill infeed pass 1/1 )
G01 Z0.10000
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G01 F200.00000
G01 X2.09064 Y34.89830
G01 X2.09064 Y36.22832
G01 X3.42066 Y36.22832
G04 P0 ( dwell for no time -- G64 should not smooth over this point )
G00 Z0.10000 ( retract )

Is this a bug, or am I missing something?

eyal0, 2023-07-13:
Also send the command line that I should use for running it.  Thanks!

hustav, 2023-07-14:
Sure. I am including the config file. Thanks!

[Archive.zip](https://github.com/pcb2gcode/pcb2gcode/files/12046439/Archive.zip)

hustav, 2023-08-05:
Hi,

Any luck reproducing this issue? BTW, setting mill-speed as a number in the range [0, 1) makes pcb2gcode generate a command G00 S0. Many laser engravers use a number in the range [0, 1] to set the laser power between 0% and 100%.

Thanks!

eyal0, 2023-08-18:
I see what you're talking about.  I'll take a look right now.

eyal0, 2023-08-18:
Seems like it's sufficient that zwork is zero for this failure to occur....

eyal0, 2023-08-18:
I have a fix.  Can I add your `Archive.zip` file into the source code as part of the test cases?  This helps ensure that the fix is regularly tested and saves me the trouble of having to make a test case.  Let me know, thanks!

hustav, 2023-08-21:
Yes, feel free to include any files you need.

eyal0, 2023-08-21:
Moved the new issue to #677

closed 2023-08-21 by eyal0