pcb2gcode/pcb2gcode #84: Copper fill converted to disconnected islands for each pad.

oyvindkinsey, 2018-01-19:
I have a gerber file created using easyeda.com, where GND is connected using a copper fill. When using `gerbv` to display this, everything looks as expected.

When I run this through `pcb2gcode`, it fails to recognize that there's a copper fill, though it does recognize the pads. As a result, it routes the pads as isolated islands instead of leaving them be as part of a contiguous copper fill.

Is this this an issue with the EasyEDA output, or with `pcb2gcode`s parsing? While `gerbv` seems to get the right idea, which might validate the EasyEDA output, maybe EasyEDA is doing something which while valid, is unorthodox and so throws off `pcb2gcode`?

EasyEDA output:

[Gerber_drill.GTL.txt](https://github.com/pcb2gcode/pcb2gcode/files/1644949/Gerber_drill.GTL.txt)

gerbv output:
[image]

pcb2gcode output:
[image]

oyvindkinsey, 2018-01-19:
After reading up on the gerber spec, and parsing through the file, it looks like the issue I'm experiencing is that `pcb2gcode` does not recognize the main fill, which is drawn by using `LPD` to draw the main polygon as dark, followed by `LPC` to draw all the 'cutouts'. 

What is the suggestion resolution here? Is there a workaround, maybe a different strategy for creating the pour, where I can run a conversion before processing the gerber file?

oyvindkinsey, 2018-01-23:
@eyal0, you seem to be actively working on this - do you know what the issue might be? If there's an issue with pcb2gcode, then I might be interested in contributing a fix if needed. While I've looked at the code, it seems like it could be easier to ramp up with some documentation?

eyal0, 2018-01-23:
Sounds like the gerberimporter isn't doing the right thing. The
gerberimporter reads in the gerber files and converts them to an internal
representation, depending on whether you're using vectorial mode.

Are you using vectorial mode? You should.

I'm not familiar with the code in that part, nor am I familiar with the
gerber spec. If you can find the problem, go for it!

oyvindkinsey, 2018-01-23:
@eyal0, you helped me find the issue - and it's actually with vectorial. With it off, the output is the expected; with it on, it's missing the copper layer. When this happens, you can see that the svg's and png's don't match up.

eyal0, 2018-01-25:
vectorial and non vectorial work differently.  I'll explain:

Non-vectorial uses a native function in gerbv to convert the input gerber to a black and white image.  That image is then worked on to convert it to toolpaths (basically tracing the path of contiguous regions).

The vectorial version uses gerbv to extract all the shapes and then converts those into boost geometry shapes and then the outline of those shapes is used for milling.

So the non-vectorial way has very simple parsing of the gerber but converting that to paths is more complicated.  Vectorial has complicated parsing of the gerber but then converting that to paths is simple.

I will look at your files and see which shapes are actually found in vectorial mode.  You're right that the later shapes are supposed to overwrite the shapes underneath and maybe that logic is broken.

eyal0, 2018-02-01:
I'd like to work on this.  Can you help?  Is it possible to take the input file and distill it to just the smallest example that demonstrates the issue?  That will save time.

eyal0, 2018-02-01:
The issue is like you said: The parser isn't handling polarity clear properly.  `gerberimporter.cpp` is quite complicated and originally written by @corna.  It's so large that I'm afraid to touch it!

Perhaps as a work around you could find a tool that can clean up the gerber?  Sorry!

corna, 2018-02-01:
Try to replace [this line](https://github.com/pcb2gcode/pcb2gcode/blob/master/gerberimporter.cpp#L657) with

```
if (polarity == GERBV_POLARITY_DARK)
    bg::union_(buffered_mls, *draws, *temp_mpoly);
else if (polarity == GERBV_POLARITY_CLEAR)
    bg::difference(buffered_mls, *draws, *temp_mpoly);
else
    unsupported_polarity_throw_exception();
```

corna, 2018-02-01:
It's wrong, ignore it.

oyvindkinsey, 2018-02-01:
Here's a the smallest sample I could think of.

[Gerber_drill.GBL.txt](https://github.com/pcb2gcode/pcb2gcode/files/1686617/Gerber_drill.GBL.txt)

corna, 2018-02-02:
Fixed in 07f4adc5a2a111929c54511b0f7e1ecae7a31bd2

closed 2018-02-02 by corna

oyvindkinsey, 2018-02-02:
That's awesome! Thanks @corna and @eyal0!