pcb2gcode/pcb2gcode #85: Feature Request :: Variable delay (G4 Pn) or pause/manual-cycle-continue (M0) after starting spindle (M3)

WebMaka, 2018-01-21:
My CNC mill uses a 3/4 HP DC motor for its spindle, which is operated by a motor controller board that's linked to my CNC controller PC. It's beastly (it can cut mild steel) but accurate enough to cut 8-mil traces. Problem is, it takes about 5 seconds to spin up.

I've been manually adding M0 (pause) lines after M3s, so program operation stops until the spindle can get up to speed. It'd sure be nice to have the option to automagically add them (or a settable delay, which is "G4 Px" where the "x" is the delay time in seconds).

corna, 2018-02-03:
It was never merged, can you download [corna/pcb2gcode](https://github.com/corna/pcb2gcode/tree/master) and check if it works?

eyal0, 2018-04-02:
@WebMaka Is this fixed now?  I see spinup-time and spindown-time options available now.

WebMaka, 2018-04-02:
I'll have to grab the latest and give it a try - haven't had time to do anything with it lately due to meatworld work overload. :-(

eyal0, 2018-04-29:
It's supposed it to work and it looks okay in testing.  I'm going to close this.  You can reopen if you find that it's not working.

closed 2018-04-29 by eyal0