pcb2gcode/pcb2gcode #87: mirror_absolute does not mirror absolute

maknoll, 2018-01-25:
Hi,

If i am using 

    mirror-absolute=true
    zero-start=false

i expect the backside to be mirrored on the on the y-axis of the gerber, instead it is mirrored at the bounding box.

https://github.com/pcb2gcode/pcb2gcode/blob/538707c0980cadd1cc766084fb3d94d9239ce1eb/surface_vectorial.cpp#L114-L116

I made a dirty fix and changed it to

     const coordinate_type mirror_axis = mill->mirror_absolute ? 
     0 : 
     ((bounding_box.min_corner().x() + bounding_box.max_corner().x()) / 2); 

Now it does exactly what i want it to do, but will surely mess up if `zero-start=true`

the following image shows the back correctly mirrored with the fix applied

[image]

Thanks for providing me awesome tools.

-Martin

maknoll, 2018-01-29:
see PR #90

eyal0, 2018-03-27:
I think that I've solved this with https://github.com/pcb2gcode/pcb2gcode/pull/110 .

Please check and let me know.  You can also merge the latest code into your PR: https://github.com/pcb2gcode/pcb2gcode/pull/91

Thanks!

eyal0, 2018-03-29:
Now with mirror-axis option, you should have all the flexibility needed to make nice, 2-sided boards.  Please re-open this issue if you have any questions.  Thanks!

closed 2018-03-29 by eyal0