pcb2gcode/pcb2gcode #88: Ubuntu installation (make) issue

d-digiacomo, 2018-01-28:
I have freshly installed Ubuntu (lubuntu-16.04.3-desktop-amd64 specifically) and am attempting to install pcb2gcode using the commands below as outlined in README.md:

`sudo apt-get update`
`sudo apt-get install build-essential automake autoconf autoconf-archive libtool libboost-program-options-dev libgtkmm-2.4-dev gerbv git`
`git clone https://github.com/pcb2gcode/pcb2gcode.git`
`cd pcb2gcode`
`autoreconf -fvi`
`./configure`
`make`
`sudo make install`

However `make` is not completing successfully - see output below. I was able to install using the above on several other PCs running the same OS last year. Perhaps an update to one of the packages installed is causing the issue? I also did try installing Boost v1.58 but could not run the last command (`./configure --with-boost=~/Downloads/boost_1_58_0 --enable-static-boost`) - ./configure could not be found. Deleting the pcb2gcode directory and running the commands above again made no difference.

[image]

d-digiacomo, 2018-01-28:
Installing using the .tar.gz archives (https://github.com/pcb2gcode/pcb2gcodeGUI/archive/v1.3.2-1.tar.gz, https://github.com/pcb2gcode/pcb2gcode/releases/download/v1.3.2/pcb2gcode-1.3.2.tar.gz) instead of git worked.

eyal0, 2018-01-30:
If `./configure` cannot be found, it's because autoreconf needs to be run.

As for the `manhattan_area_type`, maybe you're using an old version of boost geometry?

corna, 2018-02-02:
I can reproduce this in a clean docker image (ubuntu:16.04), I'll have a look.

corna, 2018-02-02:
@eyal0 538707c0980cadd1cc766084fb3d94d9239ce1eb introduced this bug, can you have a look and try to understand why it doesn't work on Boost 1.58?

I see that in Travis you switched from Boost 1.56 to 1.65: the minimum target version of Boost in pcb2gcode is 1.56 and, unless there's a strong need for a newer one, I'd like to keep that. Having 1.56 on Travis helps us to find these kind of bugs during the development.

corna, 2018-02-03:
@d-digiacomo It was a bug in Boost Polygon < 1.60, can you download [corna/pcb2gcode/boost_polygon_fix](https://github.com/corna/pcb2gcode/tree/boost_polygon_fix) and check if it works?

eyal0, 2018-02-03:
@corna I had to upgrade boost because eulerian paths uses sgementize and segmentize didn't work in older versions of boost.  If you try an older version of the code with eulerian paths, it doesn't compile.

Do you have a different suggestion?

eyal0, 2018-02-03:
I see that the intersect_segments function existed as far back as 1.56: http://www.boost.org/doc/libs/1_56_0/libs/polygon/doc/gtl_segment_concept.htm

With your fix corna/pcb2gcode/boost_polygon_fix , does it work with 1.56?  If so, great!  Dial the version back.

d-digiacomo, 2018-02-04:
@corna Boost wasn't included in a fresh install of lubuntu-16.04.3-desktop-amd64. Running `sudo apt-get install build-essential ... git` installed [Boost v1.58](https://launchpad.net/ubuntu/xenial/+package/libboost-program-options-dev). [corna/pcb2gcode/boost_polygon_fix](https://github.com/corna/pcb2gcode/tree/boost_polygon_fix) compiled without any errors using this version. Thanks.

corna, 2018-02-04:
@eyal0 
The only issue is `make check`, probably due to the fact that in 1.56 it wasn't possible to initialize a geometry object with an initialization list (only starting from 1.59). Can you fix it? [Here](https://travis-ci.org/corna/pcb2gcode/jobs/337231368)'s the log.

eyal0, 2018-02-05:
@corna Of course.

https://github.com/corna/pcb2gcode/pull/3/commits

corna, 2018-02-05:
> BTW, what do you think about putting the pre-built boost archive boost_1_56_0_amd64.tar.xz and boost_1_65_1_amd64.tar.xz on github instead of sourceforge? It's a bit odd that the github build uses sourceforge. You can attach files to releases.

I know, but they're not much useful for the users, they're used only by Travis; I think it would only be confusing if we start putting them in the Release page.

Thanks for the PR, I've merged all the changes in the main repo.

closed 2018-02-05 by corna

eyal0, 2018-02-05:
Yes, it would be confusing to have boost in the releases.  github has large-file storage but I found it too complicated for this purpose.