pcb2gcode/pcb2gcode #89: Intermediate images left on disk

ryansturmer, 2018-01-29:
It looks like there are a bunch of intermediate files generated when pcb2gcode is run - is there a way to get pcb2gcode to tidy up after itself, or to generate those files in a system-designated temporary directory instead?

corna, 2018-02-02:
With _--output-dir_ you can specify the folder where all the images will be created.

corna, 2018-02-02:
But in that case also the output gcode files would be placed in the temporary folder, which is not good
I can add an option like _--no-img_ I suppose

eyal0, 2018-02-03:
Does it make sense to add yet another option for just this, however?  The
svg files are created in many places so the solution might be mess in code.

But the problem could be solved by just typing:

`pcb2gcode && rm -f *.svg`

That seems like a good enough solution...

corna, 2018-02-04:
@eyal0 Valid point

eyal0, 2018-02-21:
@ryansturmer Will the solution mentioned above suffice?  Just `rm -f *.svg` after each run.

eyal0, 2018-03-11:
Re-open this if it's still a problem.

closed 2018-03-11 by eyal0