pcb2gcode/pcb2gcode #97: Stupid question - how to align for double sided?

hedefalk, 2018-02-07:
Sorry to post this as an issue, but I'm having trouble to understand how to align for double sided pcbs. Is there a reference anywhere for this? So far I've only used pcb2gcode for drilling and done toner transfer afterwards, but I would like to try to mill too…

eyal0, 2018-02-18:
If you have a jig that can reliably hold your board, you can flip your board over and put it back in the jig.  You'll need to make sure that the left edge of the board stays in the same place.  That's with `mirror-absolute=false`.

Alternatively, you can drill a tiny hole through your PCB in an unused part of it and make sure that you zero on that point both before and after you flip.  That is the `mirror-absolute=true` mode.

Hope that makes sense!

hedefalk, 2018-02-18:
>Alternatively, you can drill a tiny hole through your PCB in an unused part of it and make sure that you zero on that point both before and after you flip. That is the mirror-absolute=true mode.

This one I kindof understand, but that means I would need to double the space needed on the axis orthogonal to the one I flip, right? At least if my zero is in the corner? But maybe putting zero on the middle somewhere and flipping using that would work… Zero have been in the corner for me though and I have limited space (tiny t8 cnc, around 10*15cm)

Also, I'd need two holes or I would have rotation?

>If you have a jig that can reliably hold your board, you can flip your board over and put it back in the jig. You'll need to make sure that the left edge of the board stays in the same place. That's with mirror-absolute=false.

This one I still don't understand, I feel thick as a brick. I've typically worked with a piece that's bigger than the pcb that's to be milled out. If I was to flip it over x axis keeping left the same I would have bottom left corner where top left was before. Then it would only work if top margin is exactly the same as bottom margin? I have no way to do that… Is it implicit that the pcb is to be in the exact middle of the raw pcb material?

I know FlatCam had this thing where you assign two holes and choose an axis and it generates mirrors for these. That one I understand, but it would also take a bit of extra space…

hedefalk, 2018-02-18:
>I know FlatCam had this thing where you assign two holes and choose an axis and it generates mirrors for these. That one I understand, but it would also take a bit of extra space…

I take that back, no extra space, that one I understand finally :)

eyal0, 2018-02-18:
I've never made a double-sided board so I'm not an expert.  If I had to, I think that I would use `mirror-absolute=true` and then I would drill a small hole through the board.  I would place my board on the CNC for the back, line the hole up with the milling bit, and then zero the CNC.  Then I'd run back.ngc.

After that, i would flip the board over like turning a page in a book, place it on the CNC, and move the bit to be above that hole again.  I'd zero the CNC and then run front.ngc.

I think that would work.

The hole can be wherever you want.  If you want, line up the hole and instead of zeroing to 0,0, set it to -10,-10 or some other coordinate.  Then when you flip it over, do the same coordinate.  So long as the offset is the same, it should line up properly.  I think.

oyvindkinsey, 2018-03-14:
@eyal0, your approach does work, since pcb2gcode in this case with render negative X coordinates for the backside. Unfortunately, it's really not that easy to zero in on a hole with good accuracy (lest you e.g use a 1/8 bit to zero in on an 1/8 hole), and pretty much impossible if you want to use the boards 0,0 due to limited margins (e.g, I routinely work on 100mm/70mm boards with no cutouts).

The alternative supported by pcp2gcode is to mirror along the center of the Y axis, but since there's absolutely no way to identify where pcb2gode actually puts that, this is unfortunately pretty useless. If this could be identified, we could add extra gcode to add the alignment holes along this axis.

As it is, the only way I know how to identify where pcb2code puts the center line is to generate drill files using both the back and front option, pick an X position from both, and find the center.

For example, front(G0 X48.66780 Y61.36780), and back(G0 X53.74780 Y61.36780) gives me a center X of 51.2078, and I can then just add some manual gcode to put the alignment holes on this axis wherever there's room. 

Since pcb2gcode already computes this X internally (to mirror around), it would be ideal if it could also accept a list of Y positions where it would place the alignment holes along this axis. pcb2godeGUI could then render that X axis, and allow the user to click along that axis where they want the alignment holes.

oyvindkinsey, 2018-03-14:
More realistically though, I'll use the method above to identify the total width of the board as seen by pcb2gcode (just add the two numbers), and then offset holes at min(X) + offset and max(X) - offset since I typically have room in the corners (and can leverage the same holes for mounting);

hedefalk, 2018-03-14:
>For example, front(G0 X48.66780 Y61.36780), and back(G0 X53.74780 Y61.36780) gives me a center X of 51.2078, and I can then just add some manual gcode to put the alignment holes on this axis wherever there's room.

This is the first time I understood how I could even approach making double sided boards with psb2gcode. Thanks a lot! I don't know if I suck at reading docs, but I didn't even know it was mirroring on y, thought it was x like FlatCAM.

> I've never made a double-sided board so I'm not an expert. If I had to, I think that I would use mirror-absolute=true and then I would drill a small hole through the board. I would place my board on [...]

>After that, i would flip the board over like turning a page in a book, place it on the CNC, and move the bit to be above that hole again. I'd zero the CNC and then run front.ngc.

I don't know how I could possible make anything work with my hardware setup using only one hole since I don't control rotation that perfectly. I'd definitely need two alignment holes. Oh, or is that using 0,0 too as one of the alignments? Yeah, ok, I think _maybe_ I understand… However, I have this tiny CNC and don't actually have room to turn it like a page of a book. I'm doing stuff like 100*70 boards on a 150*100 machine…

oyvindkinsey, 2018-03-14:
@hedefalk, the one hole method only works if you have a jig that keeps the one axis static, e.g a set of pins that you push the board against.
Something like this can work for this, and makes it easy if you can zero your spindle perfectly.

                          ● (placed to the left of 0, only in when operating on the right side)
                             ● (placed to the right of 0, only in when operating on the left side)
●                 ●          ●                   ●

But yeah, this setup requires either twice the size, or it requires you to shift the jig and use e.g. G54 to zero in at the far right instead of far left.

hedefalk, 2018-03-14:
>But yeah, this setup requires either twice the size, or it requires you to shift the jig and use e.g. G54 to zero in at the far right instead of far left.

Yeah, I think I need to go with the two pin approach on center axis so I can use the full area of my cnc. Thx a lot for that info though!

eyal0, 2018-03-14:
I'm not a fan of the center axis because it's too hard to predict where the center will be.  For example, imagine that you first mill the front:

[image]

The purple cross is 0,0, the green are the traces, the blue is the red is the outline.  The blue line will be was pcb2gcode calculates as the "middle" axis.  It's in the middle of the milled area but it's not in the middle of the FR4.  Here's the back:

[image]

If you use set mirror-absolute to false, you need to make sure that the blue line doesn't move when you flip.  If you set mirror-absolute to true, then you need to make sure that the purple cross doesn't move.

Both of those are hard to do so I'm not really sure what the best solution is.  If you have a jig with four posts and you drill holes in the PCB so that you can flip it over reliably, then you can set 0,0 to be the bottom left post for the front side and set 0,0 to the bottom right post for the second side.  That's with `mirror-absolute=true`.

If you want mirror-absolute=false, you need to put alignment holes in your drill file.  You can find the middle of your board by yourself and then put two holes at the top and the bottom of the blue line.  The drill will drill those holes and then, when you flip the board over, you need to make sure to place the board again so that holes line up.  That seems more difficult to me because it's easier for the jig to consistently line up the FR4 than to line up the holes on the blue line.

Also, you have to calculate where the blue line is based on the shape of the board.  I'm not sure how well that code works because if you look at the example, the middle line of the outline and the middle line of the traces might not be the same thing!  I'm not sure if the mirroring code will correctly do it!

I was considering removing mirror-absolute=false and just make it always be true because I think that mirroring down the blue line is too difficult.  But I'm happy to be convinced otherwise.  Maybe try both ways and let me know which was easier?

oyvindkinsey, 2018-03-14:
@eyal0 , look at what I wrote above - there's no ambiguity as to where the center will be if you simply average out the position of any arbitrary point when seen from the left and right. 
So if you ask pcb2gcode to compute the drill holes for both the back and front, then you can average out the position of one of those holes to get the center value.

oyvindkinsey, 2018-03-14:
I'm not sure why you want your center to be in the middle of the FRC - if you want that, then you can just put you G54 at FRC center - offset. 
The point of using pins is that you simply flip the FRC on the center axis, and so in your example, the blue line will remain in place, while as the FRC will be offset to the right for the front, and to the left for the back - the main circuit will remain in dead center though.

eyal0, 2018-03-14:
Is it easy in practice to flip the board over and get the center line to
line up?

hedefalk, 2018-03-14:
>That seems more difficult to me because it's easier for the jig to consistently line up the FR4 than to line up the holes on the blue line

>Is it easy in practice to flip the board over and get the center line to
line up?

Yes, I think so by using pins. You drill two holes a bit deeper into the wasteboard and mark them. Then when you flip you use pins through the hole into the wasteboard.

I've done similar now when I've been using the cnc only for drill holes. Then aligned two papers for toner transfer using pins, it's pretty solid. I can't compare though because I don't have any jig…

oyvindkinsey, 2018-03-14:
Yeah, as @hedefalk said, you use pins that fit the holes you drilled. The more pins you put in, the less slack and the higher the precision. If you have enough wasteboard to drill 1/2", then a 1/64" pin is pretty damn sturdy and will not move on you at all.

hedefalk, 2018-03-14:
FlatCam has some info here:

http://flatcam.org/manual/procedures.html#side-pcb

They don't do it on the centerline though, they let you choose two holes yourself and then add the mirrors.

eyal0, 2018-03-17:
I had a great idea: Why not just let the user choose the mirror axis? We could make it a new parameter and it would replace all the mirror-absolute nonsense.

It would work like this: there would be a parameter called, say, mirror-axis. Whatever value you set it to,  it'll mirror along x=value.

So you first load up your FR4. Zero the machine and check that the milling and drilling will fit.

Next, use your CNC to drill alignment holes through the FR4 and into the wasteboard. Drill them on both sides of the mirror axis. For example, if you choose mirror axis=100, you can drill anywhere along x=100 or (10,10 and 190,10) or (90,10 and 110,10),etc. For every alignment hole at x,y put another at 200-x,y.

Then mill and drill like usual.

This seems like the best solution to me.

Honestly, it's all a little bit overkill because and half-competent CNC software ought to be able to do the offset for you and there's no need to add the feature to pcb2gcode. But it's easy enough to add the feature as I've described.

What do you think? Maybe offer just mirroring along whatever line x=constant . I'd like to removing mirroring along the center because, as you've noticed, you had to do some acrobatics to find the center of the board. Might as well just let users declare the center and be done with it.

oyvindkinsey, 2018-03-17:
@eyal0, that sounds like a great idea.

hedefalk, 2018-03-17:
Sounds great to me!

>Honestly, it's all a little bit overkill because and half-competent CNC software ought to be able to do the offset for you and there's no need to add the feature to pcb2gcode. But it's easy enough to add the feature as I've described.

If it's possible to do just with the cnc I'd be fine with that too. What I'm missing is just a place to start, I'm basically too stupid to make that workflow up myself I think :) Or I dunno, I think I never understood how the mirroring of pcb2gcode worked so…

Yeah, a feature would be best for me really I think but if it feels redundant/bloat from your perspective, maybe we could cook up a good wiki/tutorial-entry for beginners like me?

oyvindkinsey, 2018-03-18:
I used the approach I described earlier yesterday, and it worked swimmingly. I used a 1.5mm bit to drill a hole in each corner, and used four if the same bits as alignment pins while position the board.
Since I use double sided tape, I simply attached the tape to the board, put the bits through the holes, and then inserted then into their respective holes. Since the drill bits also acts as screws, I could 'lower' the board onto the bed with almost perfect alignment. 
Ideally I'd use proper pins, since the bits do flex a little, but I'm pretty satisfied with my first ever double sided board - it's even got electroplated though-holes!

[image]
[image]

oyvindkinsey, 2018-03-19:
Thanks @hedefalk - I'm planning to do some write-ups on this, just falling short on time right now.

I'm using 30° vbit for isolation, which pretty much makes auto-leveling mandatory, lest you end up with  widely varying widths. For auto-leveling (and milling), I use http://chilipeppr.com/jpadie, and my touch probe is optically isolated to avoid noise.

I use a homemade conductive ink for activation of the through holes (Black Magic + as much graphene as I could mix in), and a plating bath derived from [Think and Tinker](https://www.thinktink.com/stack/volumes/voliii/consumbl/cplatmix.htm), but using polyethylene glycol as the organic additive instead of the CG CLX Makeup solution (I followed https://www.youtube.com/watch?v=KTNuTv_IQp4 for this).
I use a small adjustable LM2596 buck converter hooked up to a modded ATX supply for the plating, but since this is constant voltage and not constant current, this isn't ideal - I'll fix this in the future.

This is the actual setup I made for this:
[image]
[image]

I've just been doing some basic experimentation so far, but think I can optimize this process quite a bit. I'll be mixing up some tinnit tomorrow to tin the copper too.

Next on my list is to actually start doing some SMD designs, and using a solder mask, which gets a bit more involved again.

hedefalk, 2018-03-20:
@oyvindkinsey Cool stuff! I really would like to try this. Just need to get all the chemicals, it's sometimes pretty hard in Sweden… I have a bench supply with constant current limit so that's at least something :)

oyvindkinsey, 2018-03-20:
Copper sulfate, hydrochloric acid, and sulfuric acid (battery acid) should all be fairly easy to get. And the polyethylene glycol is a common laxative.

There's better activation solutions than conductive in though, with electroless copper plating being much better (and easier to do well), but it requires calcium hypophosfate which isn't available in the states. It can be synthesized from other phosfates - see ericmitc's comments on http://www.instructables.com/id/Inexpensive-method-of-industrial-level-quality-PCB/ - but I don't have time to start experimenting with that right now.

hedefalk, 2018-07-31:
@oyvindkinsey 
>and my touch probe is optically isolated to avoid noise

I was wondering about this. Using a V-bit at 30°, was this a must or just nice to have? I'm trying to move forward with my cheap T8 CNC. Do you mean you used an optocoupler? Something like an 6n138 IC? (I have a few of those in my drawer for used for midi) 

Do you have any schematics you could share? At this point, even a photo of a hand drawing on a napkin would be really helpful :) I'm having trouble assembling a complete idea of all I need and since you use the same stack I'm intending (pcb2gcode + chilipeppr + grbl) I think you're my best hope :)

I'm pausing electroplating by the way and got brass rivets instead, these kinds: https://www.aliexpress.com/item/1000pcs-M0-9-2-5-Brass-copper-Hollow-Rivet-0-9mm-Brand-Double-sided-Circuit-Board/32793785604.html?spm=2114.search0104.3.8.6b881509PeTQXw&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10151_10065_10344_10068_10342_10343_10340_10341_10696_5011015_10084_5010915_10083_10618_10307_10820_10821_10303_10302_10869_10868_10059_100031_10103_10624_10623_10622_10621_10620,searchweb201603_44,ppcSwitch_5&algo_expid=064b5a83-49a6-4677-95d8-ea2ac71e4515-1&algo_pvid=064b5a83-49a6-4677-95d8-ea2ac71e4515&transAbTest=ae803_1&priceBeautifyAB=0

I really don't have the space for doing electroplating in my appartment so that will have to wait until I have a garage I guess.

I also have a question about offsets. When using gerbers that are a bit off I figured I'd just use offset in my millproject, but there's something that's not behaving as I thought, like the offset is applied before mirroring making the backside waay off. Like this:

x-offset=-100
y-offset=60
mirror-axis=0

Front:
[image]

Back:
[image]

Is it better to leave the offset in pcb2gcode and do something in chilipeppr or should I try to align the source gerbers?

Just for reference, if I don't use offset in pcb2gcode I have this:

#x-offset=-100
#y-offset=60
mirror-axis=0

[image]
[image]

eyal0, 2018-07-31:
Okay, let me explain how it works.

Looking at your last two photos: Imagine that I lay a blank 2-sided PCB on the CNC and zero the machine, then I drill some holes that are along the y axis, along the line x=0.  In the image below, I put three holes in green:

[image]

Next, I mill the board and make the traces.  Now I want to do the other side.  So I flip the board over, like a book, from right to left, and line up the green holes with where they were before:

[image]

Now I mill the other file.  Because you flipped along the line x=0, the part that you milled used to be on the right and now it's on the left.  This will work.

After you remove the PCB and flip it over, you might find it hard to line up the green holes.  I recommend that you use some sort of pins or maybe extra drill bits that have a snug fit.  Before you even start, put the drill at 0,0 and drill right through the PCB and into the waste plate below.  Then do it again at 0,50 and 0,100 and put drill bits into those holes.  You'll use those for alignment later.

This would work except you have two problems:
1. The holes are really far away from the part of the PCB that you are using.  So you need a big PCB.
2. You're milling at x=+158mm and x=-158mm, so your CNC needs to be at least 316mm wide even though your board is only 60 cm across.

Solution:
1. You need to get the origin closer to the board.  If you set `zero-start` to true (I think that's what you did), then the zero of the original gerber files is obeyed.  This might be important to you if you're using other tool that need to read those gerber files and zero from them.  But if not, set `zero-start=false` and pcb2gcode will make your milling next to the origin.  Or you can use `offset-x` and `offset-y` like you discovered.  `zero-start=false` basically calculates the bottom-left corner for the offsets.
2. The mirror-axis is too far away from the board.  You want to flip the board along it's middle, so that you don't need a CNC that is 316mm wide.  Try setting `mirror-axis=125` so that you will be flipping closer along the center of the board.  Now you need to make the green holes at (125,0), (125,50), (125,100).  Then put your alignments pins in there and flip.

You don't want to drill and alignment pin through the center of your board, that would suck!  Instead, you should put the alignment pins above and below the board area.  If you want, you can also put pins at the sides.  For example, at (100,50) and (150,50).  Those are the same distance from the x=125 line so it'll work when you flip.  Make as many alignment holes as you need to keep the board sturdy.

To make those alignment holes you'll use the CNC itself, of course.  Put a drill bit in, zero the machine, and then jog over to the alignment holes that you want to make and drill into the PCB and into the wasteboard below.  Make it deep enough so that you can put alignment pins in there later.

I hope that makes sense!

eyal0, 2018-07-31:
Oops, I got zero-start backwards.  You need to set zero-start to true

eyal0, 2018-07-31:
I wrote an explanation of it here: https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Two-sided-PCB-and-offsets

hedefalk, 2018-07-31:
@eyal0 Thx! zero-start was the missing piece. I had:

```
# This seems to set the origin differently, but I don't understand the docs
zero-start=false
```
:)

Now testing with:
```
zero-start=true
x-offset=5
y-offset=-110
mirror-axis=25
```

I have:

[image]
and
[image]

which finally will fit my cnc.

Just one more question - I'm gonna try to pin using existing holes. The mirror-axis param is this then not in regard to the original gerbers but the output gcode is it? I mean, if I'm gonna look for two holes that are aligned on same x, I should look at a previous run on the front.ngc? Or can I look at the gerbers? I can't do that, right since this zero-start that makes an auto offset?

eyal0, 2018-07-31:
Oh wait, a better way to look for the holes on the mirror-axis is to look at the gcode after it is generated, not the gerbers.  It should be easy to read the gcode for the drilling and look for holes where the X axis is the same.  So that would give you the mirror-axis.  Assuming that you are drilling from the front and not the back (that's an option that you can choose), then the mirroring moves the back around while the front and the drill stay put.  So you could use the value from the drilling.

But I still prefer the corners for the aforementioned reasons.

hedefalk, 2018-07-31:
Hm, I guess the mirror-axis interpretation depends on just this zero-start? Testing a bit I actually found that it might be better for me to user zero-start=false and manually offset. Because that seem to let me set the mirror-axis using gerber coordinates. I think. I have:

```
zero-start=false
x-offset=-105
y-offset=60
mirror-axis=134
```

which gives me

[image]
[image]

and the the mirror axis is chosen by picking a suitable x from the gerbers, not gcode. I don't think I could do that with mirror-axis=true.

Sorry all for putting all my basic exploration here on an issue… :)

hedefalk, 2018-07-31:
@eyal0 Oh, sorry, I didn't see you comments before posting this last one myself. Reading up now, just heads up if it didn't make sense…

hedefalk, 2018-07-31:
Yes, okay, by testing I came to the conclusion what you just wrote :)

And you're probably right about the alignment holes too… I just want to minimize any clicketiclick manual work on the grbl sender software because I suck at remembering buttons and stuff. I'd rather want to script as much as possible. Might actually try do some sed script to add those alignment holes to the drill gcode instead. I'll report back! Thanks for all the help!

oyvindkinsey, 2018-07-31:
@hedefalk, by far, the easiest thing to do is what I described in https://github.com/pcb2gcode/pcb2gcode/issues/97#issuecomment-372937154 - it can also be scripted. It relies only on you resolving the actually rendered width/2 of the gerber, eg by averaging the x of a known location from both sides.

hedefalk, 2018-07-31:
@oyvindkinsey @eyal0 

I think it might be easiest for me to just add the holes to the original excellon file so I tried that by adding a specific drill

T9C0.030

And just two holes

T9
X5.200Y-2.5
X5.200Y-6.4

This gives me a drill file looking like this:

[image]

So then I tried using mirror-axis=132.08 which is 5.2 inch in mm (this excellon was in inch)

That gives me ngc files looking just right for front and back:
[image]

[image]

However, the drill gcode is really off, it doesn't seem to respect the offset?

[image]

Is this intentional or a bug?

oyvindkinsey, 2018-07-31:
@hedefalk , if it's not working, then it hardly qualifies as the 'easiest' way :)

The method I described is very simple, and simply requires you to run pcb2gcode twice in order to identify the correct x/2 offset, which is where you then place two additional holes (super easy gcode) directly in the gerber.
No fancy setup required in chilipeppr, all default settings.

hedefalk, 2018-07-31:
@oyvindkinsey 

> if it's not working, then it hardly qualifies as the 'easiest' way :)
Yeah, you're right there 👍 

>by averaging the x of a known location from both sides.
Not sure I follow entirely again :) Wasn't that part solved with the new mirror-axis param I'm using here? My current problem is actually just that the drill gcode is offset compared to the front milling.

eyal0, 2018-07-31:
That drill file looks like it's due to a bug. I'll investigate. What settings did you use for mirror axis, drill side, etc that caused that. I did a lot of testing when I worked on the mirror axis code but I may have missed a case!

The averaging method means that you run pcb2gcode twice, once drilling on the front and once on the back. Then you look at the average and that will tell you where the mirror axis is.

However with the mirror axis option, you're not supposed to need that. Instead of looking at the file to find the mirror axis, I hoped that people could simply choose whatever mirror axis they want! And if you use the averaging method then the answer that you get will basically be the same as the mirror axis anyway!

hedefalk, 2018-07-31:
@eyal0 My entire millproject for that behaviour:

```
# options are here: https://github.com/pcb2gcode/pcb2gcode/blob/master/options.cpp#L215
#zero-start=true
#x-offset=5
#y-offset=-110

zero-start=false
x-offset=-105
y-offset=60
mirror-axis=132.08

metric=true
metricoutput=true
vectorial=true

#autolevel
#al-front=true
#al-back=true
#software=custom # no grbl?
#al-x=10
#al-y=10
#al-probefeed=20

drills-available=0.2,0.3,0.4,0.5,0.6,0.8,2.0,3.0
# https://github.com/winder/Universal-G-Code-Sender/issues/881#issuecomment-361567185
nog81=true #Maybe grbl does support now, it seemed to work?
nog91-1=true

front=../Downloaded_gerbers/matcher-F.Cu.gtl
back=../Downloaded_gerbers/matcher-B.Cu.gbl
drill=../Downloaded_gerbers/matcher.drl

output-dir=../build

#outline=vco555-Edge_Cuts.gbr
zcut=-3
cutter-diameter=1
cut-speed=30000
cut-feed=100
cut-infeed=1

zwork=-0.1
zsafe=3
zchange=5
zdrill=-3 # drill down this far
offset=1.5 # half of tool diam
# drill speed
drill-feed=150
drill-speed=30000

#onedrill=true # use only one bit

milldrill=false
milldrill-diameter=0.6
mill-feed=60
mill-speed=30000 # don't think this is controllable on my T8?
```

The project is that jar from the other issue, but I don't think this is specific to these gerbers. It just seems that the drill gcode is not offset correctly with this combination:

```
zero-start=false
x-offset=-105
y-offset=60
mirror-axis=132.08
```

eyal0, 2018-07-31:
@hedefalk Yes, I see the error on my computer, too.  I guess that I didn't do enough testing.  Sorry.  I've got a fix for it, I'll have it ready soon.

Here's how it works now:

For my test setup, I made a PCB that has a front, back, drill, and outline.  I made the front and back be the same file so that I can look at them like a mirror image.  I also used the tile options to check those, too.  Here's a config:

```
back=multivibrator-B.Cu.gbr
front=multivibrator-B.Cu.gbr
drill=multivibrator.drl
outline=multivibrator-Edge.Cuts.gbr

mill-feed=360 in/min
mill-speed=200rotations/second
voronoi=false
zwork=-0.04inch
drill-feed=30

drill-speed=12000
milldrill=true
onedrill=true
zdrill=-1.6
cut-feed=100
cut-infeed=100

cut-speed=10000
cutter-diameter=0.8
milldrill-diameter=0.8
zbridges=-0.6000
zcut=-0.13
fill-outline=true
zsafe=0.08
zchange=1.0

cut-side=back
drill-side=back
zero-start=false
x-offset=10 mm
y-offset=-0.1 inch
tile-x=2
tile-y=3
mirror-axis=10mm
```

And here's a photo:

[image]

You can see, because I didn't use zero-start, my milling is far away from the origin.  Now I'll try this:

```
cut-side=back
drill-side=back
zero-start=true
x-offset=0
y-offset=0
tile-x=2
tile-y=3
mirror-axis=0
```

[image]

You can see that the milling and drilling is closer to the origin.  Also, you can see that the right and left side are mirrors of each other and the center is 0 (mirror-axis).  Now I'll try to change the mirror-axis:

```
mirror-axis=10mm
```

[image]

Now the center line is at x=10mm and you can see that in the picture.  The moves for drills are nice, too, they are aligned properly.  Now I try to drill and cut on the front side:

```
cut-side=front
drill-side=front
```

[image]

Now the drills and outline are on the front side and I can see that they are in the right place.  And the mirror axis stays at 10mm.

So whatever value you choose for mirror-axis, that is where you put alignment pins.  Or you can put them not on the line so long as it is symmetric to the line.

Does that make sense?  I tried to explain it in the wiki but maybe I didn't do a great job!  https://github.com/pcb2gcode/pcb2gcode/wiki/Options:-Two-sided-PCB-and-offsets

If you have a better idea about how pcb2gcode can do two-sided boards without making it too complicated in the options, let me know!

hedefalk, 2018-08-01:
@eyal0 The fix works fine for me. Thx for super quick turnaround!

closed 2018-08-01 by hedefalk

hedefalk, 2018-08-01:
@eyal0 Just one thing to note, this is mostly note to self, but for the workflow of doing the alignment holes in the gerbers/excellon source files and offsetting oneself, one need to compensate the offset when selecting the rotation vector, since the mirror-axis is post offset I guess? However, I'm pretty sure I didn't do this before.

Mirror axis is 132.08 in the gerbers, this I could use as is before #175 I think. Now I need to add x-offset:

```
zero-start=false
x-offset=-105
y-offset=62
#mirror-axis=132.08
mirror-axis=27.08 # 132.08-105
```

eyal0, 2018-08-01:
I changed how the mirror axis works so that it.is after offsets and zero start. Now, how it works is that the mirror axis is based on the position on the machine. So if you choose a mirror axis that is half the width of the CNC, you'll always flip along the centerline of the machine no matter the zero start or offset. Also because the explanation of @oyvindkinsey who said that he uses the averaging method. As it now is, the averaging method answer will always be the same as the mirror axis. Essentially, there is no need for averaging anymore because it will always come out the same as the mirror axis.

Do you think that that makes more sense that way or less?